mvp/uhubctl #4: D-Link DUB-H7 silver edition not detected on Mac

wkoszek, 2016-08-15:
Hi

The back sticker of my hub says: DUB-H7. H/W Ver: A3.

I have an issue with silver edition of this hub. First of all, when I apply 5V power from an attached power supply, the hub seems to not work. When I plug the power off, devices are suddenly detected fine. (?) In other words, it seems like the hub is self-powered? Wondering if you use it the same way. 

```
$ uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.
```

The same also when I plug the D-link in.

USB hierarchy on my Mac:

```
USB:

    USB 3.0 Bus:

      Capacity: 2.02 GB (2,021,654,528 bytes)
      Removable Media: Yes
      BSD Name: disk2
      Host Controller Driver: AppleUSBXHCIWPT
      Partition Map Type: MBR (Master Boot Record)
      PCI Device ID: 0x9cb1
      PCI Revision ID: 0x0003
      PCI Vendor ID: 0x8086
      Volumes:
        6509990502:
          Capacity: 2.02 GB (2,021,538,304 bytes)
          Available: 1.23 GB (1,227,030,528 bytes)
          Writable: Yes
          File System: MS-DOS FAT16
          BSD Name: disk2s1
          Mount Point: /Volumes/6509990502
          Content: DOS_FAT_16

        Card Reader:

          Product ID: 0x8406
          Vendor ID: 0x05ac  (Apple Inc.)
          Version: 8.20
          Serial Number: 000000000820
          Speed: Up to 5 Gb/sec
          Manufacturer: Apple
          Location ID: 0x14600000 / 23
          Current Available (mA): 1800
          Current Required (mA): 224
          Extra Operating Current (mA): 0
          Capacity: 2.02 GB (2,021,654,528 bytes)
          Removable Media: Yes
          Detachable Drive: Yes
          BSD Name: disk2
          Built-In: Yes
          Partition Map Type: MBR (Master Boot Record)
          Volumes:
            6509990502:
              Capacity: 2.02 GB (2,021,538,304 bytes)
              Available: 1.23 GB (1,227,030,528 bytes)
              Writable: Yes
              File System: MS-DOS FAT16
              BSD Name: disk2s1
              Mount Point: /Volumes/6509990502
              Content: DOS_FAT_16

        BRCM20702 Hub:

          Product ID: 0x4500
          Vendor ID: 0x0a5c  (Broadcom Corp.)
          Version: 1.00
          Speed: Up to 12 Mb/sec
          Manufacturer: Apple Inc.
          Location ID: 0x14300000 / 2
          Current Available (mA): 1000
          Current Required (mA): 94
          Extra Operating Current (mA): 0
          Built-In: Yes

            Bluetooth USB Host Controller:

              Product ID: 0x828f
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 1.18
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0x14330000 / 5
              Current Available (mA): 1000
              Current Required (mA): 0
              Extra Operating Current (mA): 0
              Built-In: Yes

        Hub:

          Product ID: 0x1521
          Vendor ID: 0x04cc  (Philips Semiconductors)
          Version: 2.00
          Speed: Up to 480 Mb/sec
          Location ID: 0x14200000 / 26
          Current Available (mA): 1000
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            Nexus 7:

              Product ID: 0x4ee2
              Vendor ID: 0x18d1  (Google Inc.)
              Version: 2.28
              Serial Number: 0a83b2b0
              Speed: Up to 480 Mb/sec
              Manufacturer: asus
              Location ID: 0x14250000 / 28
              Current Available (mA): 1000
              Current Required (mA): 500
              Extra Operating Current (mA): 0

            iPhone:

              Product ID: 0x12a8
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 6.01
              Serial Number: ba8b31ea427d8730fa2412906d9df1f6e90a1d94
              Speed: Up to 480 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0x14210000 / 27
              Current Available (mA): 1000
              Current Required (mA): 500
              Extra Operating Current (mA): 0
              Sleep current (mA): 500

    USB 2.0 Bus:

      Host Controller Driver: AppleUSBEHCIPI7C9X440SL
      PCI Device ID: 0x400f
      PCI Revision ID: 0x0003
      PCI Vendor ID: 0x12d8

        Hub:

          Product ID: 0x9127
          Vendor ID: 0x05ac  (Apple Inc.)
          Version: 1.00
          Speed: Up to 480 Mb/sec
          Location ID: 0x40100000 / 1
          Current Available (mA): 1000
          Current Required (mA): 100
          Extra Operating Current (mA): 0
          Built-In: Yes

            FaceTime HD Camera (Display):

              Product ID: 0x1112
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 71.60
              Serial Number: CC2E9T04VPDJ9FLP
              Speed: Up to 480 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0x40150000 / 7
              Current Available (mA): 1000
              Current Required (mA): 500
              Extra Operating Current (mA): 0
              Built-In: Yes

            Apple Thunderbolt Display:

              Product ID: 0x9227
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 1.39
              Serial Number: 182A0933
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0x40170000 / 3
              Current Available (mA): 1000
              Current Required (mA): 2
              Extra Operating Current (mA): 0
              Built-In: Yes

            Display Audio:

              Product ID: 0x1107
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 2.09
              Serial Number: 182A0933
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0x40140000 / 6
              Current Available (mA): 1000
              Current Required (mA): 2
              Extra Operating Current (mA): 0
              Built-In: Yes
```

mvp, 2016-08-15:
This may be hardware issue with your hub. I have D-Link DUB-H7 silver edition, it has vid:pid 2001:f103, and it works just fine for me.

Of course, only hubs with external power supply can be possibly supported - self-powered hubs can't work for this purpose. There are some on-board USB hubs that do work (use `uhubctl -i` to enable scanning for those), but they draw power from motherboard.

I would recommend trying to connect this hub to Linux or Windows computer and see if you can use any USB devices attached to this hub. Another possibility is faulty power supply. You should be able to find replacement power supply on Amazon or Ebay.

At any rate, this is not an issue with `uhubctl`, but with your hardware.

closed 2016-08-15 by mvp

wkoszek, 2016-08-15:
@mvp Ok, so looks like we have a different editions of this device. I've unplugged it, captured the USB bus output, plugged it back, captured the USB bus output again, and here's the result:

```
[11:59:12][wkoszek@wkoszek_mba:~/r/autotest/www]$ ls_usb > 0
[11:59:41][wkoszek@wkoszek_mba:~/r/autotest/www]$ ls_usb > 1
[11:59:46][wkoszek@wkoszek_mba:~/r/autotest/www]$ diff -u 0 1
--- 0	2016-08-15 11:59:41.000000000 -0700
+++ 1	2016-08-15 11:59:46.000000000 -0700
@@ -74,6 +74,17 @@
               Extra Operating Current (mA): 0
               Built-In: Yes

+        Hub:
+
+          Product ID: 0x1521
+          Vendor ID: 0x04cc  (Philips Semiconductors)
+          Version: 2.00
+          Speed: Up to 480 Mb/sec
+          Location ID: 0x14200000 / 8
+          Current Available (mA): 1000
+          Current Required (mA): 0
+          Extra Operating Current (mA): 0
+
     USB 2.0 Bus:

       Host Controller Driver: AppleUSBEHCIPI7C9X440SL
```

wkoszek, 2016-08-15:
@mvp BTW, so can you tell me which hardware rev. do you have for your silver model? I'd like to get the same on eBay.

mvp, 2016-08-15:
Confusingly, my hub also has H/W ver. A3, but it has vid/pid as 2001:f103. Go figure...

wkoszek, 2016-08-15:
@mvp That sucks. I ordered 2nd one on eBay. Will let you know if it reports anything better.

mvp, 2016-08-15:
Yes, D-Link made this hardware versioning really bad. This is exacerbated by confusion of new black version, which has exact same name DUB-H7, but doesn't support selective power control...

wkoszek, 2016-08-16:
@mvp I picked up a new hub. When I connect this to the Mac, `uhubctl` doesn't show the hub at all:

```
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0100 power
   Port 2: 0100 power
   Port 3: 0100 power
   Port 4: 0103 power enable connect
   Port 5: 0503 highspeed power enable connect
   Port 6: 0000 off
   Port 7: 0103 power enable connect
```

This is just the USB hub in the screen itself.

Any hardware that is more readily available that you know would just work?

mvp, 2016-08-16:
@wkoszek, one easily available hardware that definitely works is white Apple keyboard, model A1243 (and costs $20-$50 depending where you shop.
It also works for my Apple Thunderbolt Display 27" - if you have it, try plugging into USB ports in the back of it.

wkoszek, 2016-08-16:
@mvp How is your screen identified? I also have Thunderbolt screen, but selective port switching doesn't work. Can you post VID/PID?

mvp, 2016-08-16:
```
$ uhubctl
Current status for hub 64-1.1, vendor 05ac:1006, 3 ports
   Port 1: 0100 power
   Port 2: 0303 lowspeed power enable connect
   Port 3: 0100 power
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0303 lowspeed power enable connect
   Port 3: 0100 power
   Port 4: 0103 power enable connect
   Port 5: 0503 highspeed power enable connect
   Port 6: 0100 power
   Port 7: 0103 power enable connect
```

First one in list above is keyboard, second is Apple thunderbolt display.
If I plug flash drive to display USB port closest to thunderbolt port, it is detected and mounted by Mac.
I can "eject" this drive by disabling power on its port using command like this:

    uhubctl -l 64-1 -p 3 -a 0

and re-plug it by

    uhubctl -l 64-1 -p 3 -a 1

wkoszek, 2016-08-16:
@mvp Hm:

```
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0303 lowspeed power enable connect
   Port 3: 0100 power                            <------------------------------------- this might be my port 1
   Port 4: 0103 power enable connect
   Port 5: 0503 highspeed power enable connect
   Port 6: 0100 power                           <-------------------------------------- or this
   Port 7: 0103 power enable connect
```

So looks like my `port 1` reported just as a `power` is your port  3 or 6. Can you connect more stuff (smartphone or something) and see if power from other ports also goes away?

mvp, 2016-08-16:
@wkoszek, it didn't work well at start - either flash drive or mouse were misbehaving. I have replugged power cord to the display, and everything started to work. Also, it seems that default state for port 6 is `off` - some weirdness on USB hub of this display, i guess it is not really wired anywhere.

mvp, 2016-08-16:
I have mouse and keyboard connected, so all ports are used up. Turning off flash drive does not affect mouse or keyboard.

wkoszek, 2016-08-16:
@mvp Ok. It doesn't work for me.