mvp/uhubctl #5: 27" Mac screen issues with power-cycling of ports

wkoszek, 2016-08-15:
*First of all - thanks a lot for `uhubctl`. I think it's a remedy for hanging USB in tablets and smartphones I work with*

## Setup

I have 27" Apple Retina screen hooked up to a Macbook Air. I want to use it to selectively power cycle ASUS Android tablet and an iPhone which I've hooked up to the screen.

## First note

My understanding is that port 1 is special? Output:

```
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0100 power
   Port 2: 0503 highspeed power enable connect
   Port 3: 0503 highspeed power enable connect
   Port 4: 0103 power enable connect
   Port 5: 0503 highspeed power enable connect
   Port 6: 0000 off
   Port 7: 0103 power enable connect
```

It only reports power, but not anything else? My empirical experiments show it's a special thing toggling power for the whole bus. Just wanted to know if this is documented. Below I provide the USB bus topology. This screen has 7 logical USBs, but the back of the screen has 3 physical USB ports.

## Issue

What I'm seeing: I can't power-cycle the `power2...7` if Android phone/tablet is hooked up to it. Example:

```
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0100 power
   Port 2: 0503 highspeed power enable connect <------- Android phone
   Port 3: 0100 power
   Port 4: 0103 power enable connect
   Port 5: 0503 highspeed power enable connect
   Port 6: 0100 power
   Port 7: 0103 power enable connect
```

The output:
```
[11:41:30][wkoszek@wkoszek_mba:~/r/autotest/www]$ uhubctl -p 2 -a off
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 2: 0503 highspeed power enable connect
Sent power off request
New status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 2: 0000 off
[11:41:55][wkoszek@wkoszek_mba:~/r/autotest/www]$ uhubctl -p 2 -a on
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 2: 0000 off
Sent power on request
New status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 2: 0100 power
```

So looks like it's OK from the `uhubctl` reporting point of view, but I don't see this power-cycle happening on a device. I'm looking at the battery status and "debugging" mode icon, since I'm developing on this phone. But when I do:

```
[11:43:50][wkoszek@wkoszek_mba:~/r/autotest/www]$ uhubctl -p 1 -a off
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0100 power
Sent power off request
New status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0000 off
[11:43:56][wkoszek@wkoszek_mba:~/r/autotest/www]$ uhubctl -p 1 -a on
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0000 off
Sent power on request
New status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0100 power
```

The power-cycle is reported by a device.

Now weird thing. I unplug everything and connect iPhone to port 3:

```
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 1: 0100 power
   Port 2: 0100 power
   Port 3: 0503 highspeed power enable connect <----------- iPhone.
   Port 4: 0103 power enable connect
   Port 5: 0503 highspeed power enable connect
   Port 6: 0100 power
   Port 7: 0103 power enable connect
```

When I do:

```
[11:47:20][wkoszek@wkoszek_mba:~/r/autotest/www]$ uhubctl -p 3 -a cycle
Current status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 3: 0503 highspeed power enable connect
Sent power off request
New status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 3: 0000 off
Sent power on request
New status for hub 64-1, vendor 05ac:9127, 7 ports
   Port 3: 0100 power
```

it works fine. The reporting is OK, and the phone gets power cycled. Same when I replugged it in port 2.

I'm trying to understand if other people see the same issue or whether it's an expected behaviour.

mvp, 2016-08-15:
Thank you! :)

* Note that USB ports as reported by libusb don't necessarily map well to actual physical ports. Some of them may be not even wired - e.g. uhubctl discovers 7 ports, but only 3 physical ports could be present. You may need to use some experimentation to find out which physical port is which from uhubctl (actually, libusb) point of view.

* Some USB hubs may act really weird if high-powered devices are connected. Normally, USB hubs can provide just 500mA of power, however some modern Android and iOS devices like to draw a lot more, e.g. 1A or even 2A of power. I observed that in those cases, some hubs may refuse to turn off power, or they do turn it off only to automatically reenable it back milliseconds later - this is why I added command line swtich `--repeat` - if used, it will try to turn off power many times in quick succession, and sometimes it does help to turn off stubborn device.

* Some hubs report that selective power control is supported, but apparently some manufacturers want to save few cents and actual power rails maybe not really connected, which makes uhubctl not work...

mvp, 2016-08-15:
Not much can be done to fix this hardware issue by `uhubctl` itself, closing...

closed 2016-08-15 by mvp

wkoszek, 2016-08-15:
@mvp I'm OK with closing, but how about we move what you/I wrote to the README? Also maybe it'd be worth starting a table of hubs that are known to cause issues/known to work, and how to check which is which?

mvp, 2016-08-15:
Ok, I will add short summary of hw issues into README later this week.

wkoszek, 2016-08-15:
Sounds good. You may reopen until then. Thanks, take care.