mvp/uhubctl #8: 'libusb_get_port_path' is deprecated since libusb v1.0.16

rleitner, 2017-02-20:
Since libusb version 1.0.16 the `libusb_get_port_path` function is deprecated. Instead `libusb_get_port_numbers` should be used.

For more details see https://github.com/libusb/libusb/blob/v1.0.16/ChangeLog#L9

mvp, 2017-02-20:
Originally I was using `libusb_get_port_numbers`, however this breaks on some OS, e.g. Centos 7,
so I committed this https://github.com/mvp/uhubctl/commit/6173f7db13e05a12a967d76c79d7676d661e5c86 as a fix.
So sorry, but this has to stay the way it is for now to be compatible with libusb 1.0.12.

closed 2017-02-20 by mvp

rleitner, 2017-02-20:
Would checking `LIBUSB_API_VERSION` in macros to use `libusb_get_port_numbers` only when it's available be fine?

mvp, 2017-02-20:
It would. I just didn't want to put extra `ifdef`s and used older call which is present everywhere.
If you want to provide ifdef patch, you're welcome :)