mvp/uhubctl #12: How do I determine if a hub is supported?

xxedgexx, 2017-02-24:
I’m trying to find a compatible USB hub that meets my needs.  One, I’m trying to find something with more ports than the listed compatible devices, hoping for a minimum of 10 ports.  Two, I’m looking for something that has a dedicated charging port with 2.1amps because I’m trying to power a raspberry pi using the USB hub.

I purchased this Anker hub:

https://www.amazon.com/gp/product/B00GSLMTQ8/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

uhubctl sees the hub and the ports, but I’m having no luck controlling power to individual ports.

root@raspberrypi:/usr/local/src/uhubctl# ./uhubctl -n 2109
Current status for hub 1-1.2.4.4.4, vendor 2109:2812, 4 ports
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.2.4.4, vendor 2109:2812, 4 ports
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 highspeed power suspend enable connect
Current status for hub 1-1.2.4, vendor 2109:2812, 4 ports
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 highspeed power suspend enable connect
Current status for hub 1-1.2, vendor 2109:2812, 4 ports
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 highspeed power suspend enable connect

I can toggle the power and uhubctl shows the port as off but the device connected to the port continues to operate.  My question is really how do I determine if a hub has the support I need to toggle power on individual ports and is there anything else I need to do to further test this hub to see if it will work?  Or is it flat out not compatible?  Is there something I can see from lsusb that will tell me if a hub is supported?

I also purchased one of your recommended hubs, the:

D-Link: DUB-H7

and this works just fine but I need more ports and something that can power the pi.

mvp, 2017-02-24:
Oh my, this topic deserves big article to explain :)

USB hub manufacturers typically buy USB chipsets from very few chipset makes like NEC, TI, Via, etc.

Normally, most chipsets do support full USB specification, including individual port power switching (aka PPPS - Per-Port Power Switching) - this is what uhubctl needs in order to work as expected.

However, many hub manufacturers choose to save on actual on/off circuitry and do disable PPPS on chipset level. What's worse, some don't even bother fiddling with chipset settings! In that case, any API trying to talk to USB hub (like uhubctl does) will find that PPPS seems to be supported. Sending commands to turn ports on/off would work and save state (just like what you observe), but because on/off circuitry is physically missing, nothing is going to happen...

You may also notice that Anker hub is seen not as 1 USB device, but whopping 4 daisy-chained 4-port hubs. This is typically very bad sign, in my experience such hubs never work well.

D-Link DUB-H7 is true 7 port hub, and it works well because they didn't go cheap on switching circuitry.

If you want to have more powerful hub with working PPPS, I would imagine USB3 hubs should work better, because USB3 must support at least 0.9A per port, and often 1A is fine. However, I don't know if there is any USB3 hub that actually supports PPPS properly.

Easiest way to tell is to run uhubctl and see if power is actually controlled as expected.

If you happen to find any hub that works and is not listed as supported, please report: I'd like to add it to the list.

I hope this helps :)

xxedgexx, 2017-02-24:
Thank you.  This makes sense and is exactly the behavior I observe.  Very unfortunate that manufacturers don't go all the way in terms of support.  In my case, I'm trying to control up to 10 different devices and the hub is connected to a raspberry pi.  So it would just be very convenient if there was a 10+ port hub with a dedicated charging port (typically 2.1amps which is plenty to power the pi).  This would mean the difference between having to plug three things in (the pi and two hubs) and just a single plug in to the wall.

xxedgexx, 2017-02-24:
I'm going to give this one a try:

mvp, 2017-02-24:
That looks very promising. Please report if it works! Thank you!

xxedgexx, 2017-02-27:
So unfortunately it doesn't look like this new hub works either despite the feature set claiming to support Per Port Power Switching.

root@raspberrypi:/opt/humistat/bin# ./uhubctl -n 174c
Current status for hub 1-1.3.4, vendor 174c:2074, 4 ports
   Port 1: 0000 off
   Port 2: 0000 off
   Port 3: 0000 off
   Port 4: 0000 off
Current status for hub 1-1.3.3, vendor 174c:2074, 4 ports
   Port 1: 0000 off
   Port 2: 0000 off
   Port 3: 0000 off
   Port 4: 0000 off
Current status for hub 1-1.3, vendor 174c:2074, 4 ports
   Port 1: 0000 off
   Port 2: 0000 off
   Port 3: 0507 highspeed power suspend enable connect
   Port 4: 0507 highspeed power suspend enable connect

All ports claiming to be off, yet the device I have plugged in continues to get power.  I've tried all the ports on the hub, device still runs.

Anything else I can try?  Seems strange that the manufacturer states PPPS and yet, seems to not work at all.  Any chance USB 3.0 standards do it a different way???  Any chance the current USB drivers in the kernel aren't doing the right thing?

Linux raspberrypi 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux

Thanks

mvp, 2017-02-27:
Not sure what's going on. Another user just reported that Anker ultra slim 4-port USB 3.0 data hub with 12W power adapter works great, and I added it to compatibility list. It proves that USB 3.0 hubs can work fine. But I guess you're looking for more than 4 ports?

xxedgexx, 2017-02-27:
Yes, for my needs, I'm trying to get at least 10 data ports and at least 1 charging port.  Strange because the last hub I tried was an Anker...

mvp, 2017-02-27:
This is thread about working Anker hub for reference: https://github.com/mvp/uhubctl/issues/14

xxedgexx, 2017-02-27:
Thanks.  Like I mentioned above, the D-Link DUB-H7 works fine.  I can obviously just get another one of these hubs.  Unfortunately, this now means I go from 2 power plugs to the wall, to 3.  I was trying to find something with 10 ports and a charging port which would then reduce my plugs to the wall to only one.

mvp, 2017-02-27:
We need more people to try running uhubctl across as many USB hubs as possible to compile comprehensive compatibility list. I myself started this in 2009 by going to Fry's and buying 10 different hubs that were available on a floor. Only D-Link DUB-H7 worked back then. Couple more reported PPPS, but didn't really work.

xxedgexx, 2017-02-27:
This looks like the control manufacturer for this hub I'm trying to get working:

http://www.asmedia.com.tw/eng/e_show_products.php?item=128

mvp, 2017-02-27:
Interesting point: this talks about uploadable firmware. From what I've seen, most, if not all, USB3 hubs can have firmware upgraded. Perhaps you need to upgrade yours?

xxedgexx, 2017-02-27:
Found firmware, which of course is only applicable via Windows.  Tried firing up a Windows VM on my Mac.  Windows didn't see the hub.  So I tried the hub just on my mac and noticed OS X was throwing up an error that "This USB accessory requires power", so it looks to me as if this controller is bad to begin with.  So do I try another one?  Not sure.  The vendor for this particular hub barely even has a website.  It's probably not a reliable piece of hardware.

mvp, 2017-02-27:
Another simple approach to this: grab your Mac and just go to any big computer store, like Fry's or Best buy. Try every hub on a USB shelf - run uhubctl on it and see if it's compatible. Profit!

mvp, 2017-03-09:
I don't think there is anything actionable here from uhubctl standpoint. Closing the issue. If you find new compatible hub, please report!

closed 2017-03-09 by mvp

mvp, 2017-03-16:
@phhusson reported in https://github.com/mvp/uhubctl/issues/17 that Amazon Basics USB 3.0 7 ports hub works fine, and it has high speed charging port.

Maybe this will fit your needs?

mvp, 2017-06-13:
Note that last version of uhubctl can control port 2 of onboard USB hub on some Raspberry PI models.

SlySven, 2017-06-15:
> Note that last version of uhubctl can control port 2 of onboard USB hub on some Raspberry PI models.

That is, the Model __B__ ones which are the ones with a built-in LAN9512 (earlier 2-USB port version) or LAN9514 (later 4-USB port ones) IC and that the port 2 is wired in hardware to control all the USB ports (2 & 3 or 2-5) respectively but not the wired LAN that those ICs also provide via the internal port 1 as far as I understand it.