mvp/uhubctl #14: Brand new, easy to purchase, per-port switchable hub

jstockdale, 2017-02-25:
I've been doing some testing and it appears these support per port power! Although, it appears they only enumerate ports that have devices plugged in (I'm seeing one that reports being a "3 port" hub and one that reports being a "4 port" hub on my system.

Very surprised. This is a cheap and easy-to-purchase modern, good-looking hub.

https://www.amazon.com/Anker-Ultra-Slim-4-Port-Data/dp/B00XMD7KPU

mvp, 2017-02-25:
I doubt this could work well, considering it doesn't have its own power supply. Any hub that has ability to truly power on/off its ports should have external power supply, that is 500mA for USB2, and 900mA for USB3 per every USB port.
Perhaps this is why it's called data hub?

That said, I don't understand what the issue here. Do you want to add it to supported devices list?

jstockdale, 2017-02-25:
The existing supported hubs are difficult to source, and many are buggy or out of production.

This "Data Hub" does in fact have a power input (micro USB) and the power ports are switched on and off with uhubctl.

mvp, 2017-02-25:
Ok, I will add this hub to the list. However, note that micro USB cannot possibly be proper power source for 4 port hub. For truly compliant operation, hub must have at least 10W power supply (5V, 4 ports, 500mA each), and twice that for USB3 operation, which micro USB input from your computer USB port cannot provide.

jstockdale, 2017-02-25:
Thanks! For what it's worth, I misspoke. There are two versions of this hub and I didn't realize I was looking at the one Anker ships with power supply https://www.amazon.com/Anker-4-Port-Adapter-Charger-Included/dp/B0192LPK5M – a 2.4A 5V charger. It shows up the same on the USB bus but it has a 12W power supply and micro usb input. The Data hub does not have a power input, so perhaps don't recommend / mention that one has support. Sorry for the confusion on the different models.

jstockdale, 2017-02-25:
Also, thanks so much for writing this tool! It's incredibly useful. Two thumbs up. Five stars. A+++ would recommend. :-)

jstockdale, 2017-02-25:
I'm also curious. What's the main purpose of this tool? To  switch power? Because I've been using it to turn on and off usb devices without unplugging them and it works amazing for that.

mvp, 2017-02-25:
Added this hub to compatible list: https://github.com/mvp/uhubctl/commit/59f94010d906d9daa8fa7f300d33ed67cd0a6bbb

I just wanted to be able to programmatically control USB power.
Since then, people use it for very unexpected applications.
I've seen someone to use it for sending morse code: https://gist.github.com/dardo82/52799ae762e8673c8726
Some other guys use it to reset buggy devices programmatically.
I guess you can use it to control USB fan or USB light :)

closed 2017-02-25 by mvp

bertani, 2017-03-07:
I have just tried the 10w version of this (from amazon UK) and it didn't work, bad luck.

mvp, 2017-03-07:
Perhaps there are different revisions for this hub. @jstockdale, can you tell more about hub that works? Like exact model number, or hw revision, or USB VID:PID? @bertani: can you also do the same for the hub that didn't work?

bertani, 2017-03-08:
Sure, this is **NOT** working:

Anker 4-Port Ultra Slim USB 3.0 Data Hub (10W), Model: "**A7518**"
USB VID:PID **2109:0812**
URL: https://www.amazon.co.uk/gp/product/B0192LVM86/

bertani, 2017-03-09:
As a future reference: unfortunately it seems like there is no obvious way to buy the 12W version from Europe.

jstockdale, 2017-03-09:
Yes of course. I will check when I get back to my computer -- might be tomorrow.

aciani, 2017-07-29:
I just bought the recommended Anker AK-A7518111 from Anker Direct on Amazon.  It does not have per-port power switching.  The VID:PID is 2109:2813.  This is the same VID:PID for the Inland hub that I erroneously reported.

The Anker hub should be removed from the list of supported devices.  It is definitely not working.  It also squeals when plugged into a USB 2.0 port.

I have contacted Anker.  Will see what they say.

mvp, 2017-07-29:
I often see that vendors continue selling hubs with exact same model id, but completely different usb chipsets. One example is D-Link DUB-H7: old silver version works, new black has exact same model name, but completely different chip and it doesn't work!

Perhaps we have similar situation with Anker? I wonder if device previously reported by @jstockdale is slightly different hw revision.

aciani, 2017-08-05:
An update on the 2109:2813.  This controller is a VIA Labs VL813.  I finally found a datasheet for it, and surprise, it only has two power-enable control lines.  The pin USBHPE1 is Power Enable for Gangmode Ports.  The pin USBHPE2B is Power Enable for Charging Ports.

The VL813 supports something called High-speed Charging, which allows devices to draw upwards of 1500 mA.  The pin USBHPE2B is used to cut power to specially designated charging ports, in the event a device tries to draw too much power.  Neither the Inland or Anker hubs have charging ports.

For some reason, uhubctl sees this hub controller as supporting per-port switching.  This controller cannot support per-port switching.  It can only switch two groups of ports, ganged normal ports and ganged charging ports, assuming the manufacturer included the necessary transistors.

mvp, 2017-08-06:
uhubctl detects it as supporting PPPS because hub reports this. I guess this is incorrect hub configuration. This hub was removed from compatibility list anyway.

jstockdale, 2017-08-06:
sorry guys. :( the version i had seemed to work but perhaps i didn't test carefully enough.

aciani, 2017-08-06:
jstockdale, your version may have worked.  As mvp noted, manufacturers may change the insides, but keep the outsides and model number.

You never mentioned the VID:PID for yours.  bertani reported 2109:0812.  This is a VL812, and like the VL813, it can only support ganged switching, assuming the manufacturer added the transistors.

mvp, 2018-02-18:
@aciani, @jstockdale: Can you please try this one again using uhubctl built from master branch?
Be sure to turn off power for both USB2 and USB3 virtual hubs. Thanks!

GatorEE, 2019-06-07:
Have you tried the Acroname hubs? A bit more expensive than these options, but well powered, pretty solid and easy to use.
www.acroname.com

mvp, 2019-06-07:
@GatorEE, Acroname is using proprietary power control protocol.
`uhubctl` supports per-port power switching as defined by USB 2.0 standard (you find a copy on www.usb.org or here: http://www.poweredusb.org/pdf/usb20.pdf ), chapter `11.11 Hub Port Power Control`.
If Acroname provides specification about their control method, I wouldn't mind to code control for their hardware into `uhubctl`.

mvp, 2019-06-07:
@GatorEE, small excerpt from this chapter:

## 11.11 Hub Port Power Control
Self-powered hubs may have power switches that control delivery of power downstream facing ports but it
is not required. Bus-powered hubs are required to have power switches. A hub with power switches can
switch power to all ports as a group/gang, to each port individually, or have an arbitrary number of gangs of
one or more ports.
A hub indicates whether or not it supports power switching by the setting of the Logical Power Switching
Mode field in wHubCharacteristics. If a hub supports per-port power switching, then the power to a port is
turned on when a SetPortFeature(PORT_POWER) request is received for the port. Port power is turned off
when the port is in the Powered-off or Not Configured states.

abu-matterize, 2026-08-17:
anyone still around? i'm building [vbus.dev](https://vbus.dev), 4-port usb hub, per-port power control over network. planning to ship by feb'27. curious who's actually in, 100 is my goal. reserve now for $10. **fully refundable until it ships**

SlySven, 2026-08-20:
> anyone still around? i'm building [vbus.dev](https://vbus.dev), 4-port usb hub, per-port power control over network. planning to ship by feb'27. curious who's actually in, 100 is my goal. reserve now for $10. **fully refundable until it ships**

TBH How does that improve upon https://www.uugear.com/product/mega4-4-port-usb-3-ppps-hub-for-raspberry-pi-4b/ ?

abu-matterize, 2026-08-20:
it's a great product, tbh. but the use-case is different, [vbus.dev](https://vbus.dev) is purpose built for embedded firmware developers to flash and test their firmware run ci on real hardware, no pi in between.

mvp, 2026-08-20:
BTW, Mega4 doesn't require RPi, it works with any USB host. Considering it costs just $35, makes it hard to beat by value.

abu-matterize, 2026-08-20:
true, vbus.dev is host by default, no need for external hosts, plug in ethernet, plug your target dut (st, esp or any usb jtag) to any port, vbus cli enables power management.

erase, flash, reset (dedicated boot and reset io for dfu mode) as you do with vendor toolchains, from anywhere

mvp, 2026-08-20:
In that case, what does vbus.dev has anything to do with uhubctl? I assume you provide separate control software to your custom hardware?