mvp/uhubctl #17: Works with Amazon Basics USB3 7-ports hub

phhusson, 2017-03-16:
I don't know if the reference is stable on the long term
But I tested this on https://www.amazon.fr/dp/B00E4J7PTM/ Amazon Basics USB 3.0 7 ports hub, and power on/off works fine.

It appears as two Texas Instrument TUSB8041 hubs, IDs are 0451:8142
The sample I have has been bought in April 2015, so a refresh of the information would be useful.

mvp, 2017-03-16:
Added in https://github.com/mvp/uhubctl/commit/2e946e3c70acd503303bac34d6e39e783a8b2eda

mvp, 2017-03-16:
Picture from product page for B00E4J7PTM shows actual model ID as B00E4J3708.
Exact same looking product sold in US on amazon.com has model ID of B00E6GX4BG.
I assume all of them should work the same.

closed 2017-03-16 by mvp

xxedgexx, 2017-03-16:
If this one would work, I'd be in business:

mvp, 2017-03-16:
Give it a try and report if it works :)

xxedgexx, 2017-03-16:
Will do.  Amazon has the best return policy in the biz!

phhusson, 2017-03-16:
Though I wonder how the "fast charge" output triggers, and whether they can be switched in software from USB connection but slow charge, to no usb connection, but fast charge.

xxedgexx, 2017-03-17:
Received the 10 port version of the hub today and unfortunately uhubctl doesn't seem to recognize it:

[   75.474468] usb 1-1.3: USB disconnect, device number 4
[   75.474492] usb 1-1.3.2: USB disconnect, device number 5
[   75.476607] usb 1-1.3.3: USB disconnect, device number 6
[  152.779448] usb 1-1.2: new high-speed USB device number 7 using dwc_otg
[  152.881150] usb 1-1.2: New USB device found, idVendor=2109, idProduct=2812
[  152.881173] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  152.881187] usb 1-1.2: Product: USB2.0 Hub             
[  152.881200] usb 1-1.2: Manufacturer: VIA Labs, Inc.         
[  152.882132] hub 1-1.2:1.0: USB hub found
[  152.882398] hub 1-1.2:1.0: 4 ports detected
[  153.179438] usb 1-1.2.2: new high-speed USB device number 8 using dwc_otg
[  153.291061] usb 1-1.2.2: New USB device found, idVendor=2109, idProduct=2812
[  153.291080] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  153.291093] usb 1-1.2.2: Product: USB2.0 Hub             
[  153.291106] usb 1-1.2.2: Manufacturer: VIA Labs, Inc.         
[  153.292058] hub 1-1.2.2:1.0: USB hub found
[  153.292310] hub 1-1.2.2:1.0: 4 ports detected
[  153.389455] usb 1-1.2.3: new high-speed USB device number 9 using dwc_otg
[  153.501087] usb 1-1.2.3: New USB device found, idVendor=2109, idProduct=2812
[  153.501107] usb 1-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  153.501120] usb 1-1.2.3: Product: USB2.0 Hub             
[  153.501133] usb 1-1.2.3: Manufacturer: VIA Labs, Inc.         
[  153.502088] hub 1-1.2.3:1.0: USB hub found
[  153.502339] hub 1-1.2.3:1.0: 4 ports detected
[  243.432872] usb 1-1.2: USB disconnect, device number 7
[  243.432896] usb 1-1.2.2: USB disconnect, device number 8
[  243.433801] usb 1-1.2.3: USB disconnect, device number 9
[  376.290510] usb 1-1.2: new high-speed USB device number 10 using dwc_otg
[  376.392130] usb 1-1.2: New USB device found, idVendor=2109, idProduct=2812
[  376.392149] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  376.392163] usb 1-1.2: Product: USB2.0 Hub             
[  376.392176] usb 1-1.2: Manufacturer: VIA Labs, Inc.         
[  376.393129] hub 1-1.2:1.0: USB hub found
[  376.393387] hub 1-1.2:1.0: 4 ports detected
[  376.690511] usb 1-1.2.2: new high-speed USB device number 11 using dwc_otg
[  376.802229] usb 1-1.2.2: New USB device found, idVendor=2109, idProduct=2812
[  376.802253] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  376.802267] usb 1-1.2.2: Product: USB2.0 Hub             
[  376.802281] usb 1-1.2.2: Manufacturer: VIA Labs, Inc.         
[  376.803216] hub 1-1.2.2:1.0: USB hub found
[  376.803509] hub 1-1.2.2:1.0: 4 ports detected
[  376.900517] usb 1-1.2.3: new high-speed USB device number 12 using dwc_otg
[  377.012135] usb 1-1.2.3: New USB device found, idVendor=2109, idProduct=2812
[  377.012154] usb 1-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  377.012167] usb 1-1.2.3: Product: USB2.0 Hub             
[  377.012180] usb 1-1.2.3: Manufacturer: VIA Labs, Inc.         
[  377.013131] hub 1-1.2.3:1.0: USB hub found
[  377.013373] hub 1-1.2.3:1.0: 4 ports detected

root@raspberrypi:/opt/humistat/bin# ./uhubctl -l 1-1.2.2
No compatible smart hubs detected at location 1-1.2.2!
Run with -h to get usage info.
root@raspberrypi:/opt/humistat/bin# ./uhubctl -l 1-1.2.3
No compatible smart hubs detected at location 1-1.2.3!
Run with -h to get usage info.

Just the onboard USB
root@raspberrypi:/opt/humistat/bin# ./uhubctl 
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0507 highspeed power suspend enable connect
   Port 3: 0100 power
   Port 4: 0100 power
   Port 5: 0100 power

mvp, 2017-03-17:
Hub that was @phhusson reported working has vid:pid as 2109:2811, but yours is 2109:2812...
They differ just by 1 :(
Can you please slightly patch `is_smart_hub()` and see which values of lpsm and ocpm are reported by this hub? uhubctl requires that both lpsm (logical power switching mode) and ocpm (over-current protection mode) are supported per port. There is small possibility that lpsm is per port, but ocpm is ganged (all ports at once). In that case, we might be able to make it work by relaxing ocpm check.

xxedgexx, 2017-03-17:
I'd be happy to.  Can you send me the appropriate patch?  Not much of a developer but I can apply patches like nobody else :-)

mvp, 2017-03-18:
Check these lines: https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L204-L209
Try to just printf() ocpm, or remove ocpm constraint altogether and see if it helps

xxedgexx, 2017-03-18:
I added this:

--- uhubctl/uhubctl.c	2017-03-17 17:09:30.733541358 -0700
+++ uhubctl.new/uhubctl.c	2017-03-17 17:07:00.385032555 -0700
@@ -204,6 +204,7 @@
             int lpsm = uhd->wHubCharacteristics[0] & HUB_CHAR_LPSM;
             /* Over-Current Protection Mode */
             int ocpm = uhd->wHubCharacteristics[0] & HUB_CHAR_OCPM;
+            printf("OCPM %d\n", ocpm);
             /* Both LPSM and OCPM must be supported per-port: */
             if ((lpsm == HUB_CHAR_INDV_PORT_LPSM) &&
                 (ocpm == HUB_CHAR_INDV_PORT_OCPM))

and here's the output:

root@raspberrypi:/usr/local/src/uhubctl.new# ./uhubctl 
OCPM 0
OCPM 0
OCPM 0
OCPM 8
OCPM 8
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0507 highspeed power suspend enable connect
   Port 3: 0100 power
   Port 4: 0100 power
   Port 5: 0100 power

phhusson, 2017-03-18:
What about uhubctl -l 1-1.2.3? (to know which hub we're looking at)

xxedgexx, 2017-03-18:
Tried that.  It's in my response above.

root@raspberrypi:/usr/local/src/uhubctl.new# ./uhubctl -l 1-1.2.3
OCPM 0
OCPM 0
OCPM 0
OCPM 8
OCPM 8
No compatible smart hubs detected at location 1-1.2.3!
Run with -h to get usage info.

mvp, 2017-03-18:
Hmm. Reported ocpm of 8 actually looks good:
#define HUB_CHAR_INDV_PORT_OCPM 0x0008 /* Per-port over-current protection */

What about lpsm? We need it to be 1:
#define HUB_CHAR_INDV_PORT_LPSM 0x0001 /* Per-port power switching */

Unacceptable would be 2:
#define HUB_CHAR_NO_LPSM        0x0002 /* No power switching */

And 0 means ganged (all ports at once):
#define HUB_CHAR_COMMON_LPSM    0x0000 /* All ports at once power switching */

mvp, 2017-03-18:
Also, try to simply remove this block: `(ocpm == HUB_CHAR_INDV_PORT_OCPM)`

xxedgexx, 2017-03-18:
```
--- uhubctl/uhubctl.c	2017-03-17 17:09:30.733541358 -0700
+++ uhubctl.new/uhubctl.c	2017-03-17 17:26:17.135096921 -0700
@@ -204,9 +204,10 @@
             int lpsm = uhd->wHubCharacteristics[0] & HUB_CHAR_LPSM;
             /* Over-Current Protection Mode */
             int ocpm = uhd->wHubCharacteristics[0] & HUB_CHAR_OCPM;
+            printf("OCPM: %d\n", ocpm);
+            printf("LPSM: %d\n", lpsm);
             /* Both LPSM and OCPM must be supported per-port: */
-            if ((lpsm == HUB_CHAR_INDV_PORT_LPSM) &&
-                (ocpm == HUB_CHAR_INDV_PORT_OCPM))
+            if (lpsm == HUB_CHAR_INDV_PORT_LPSM)
             {
                 rc = uhd->bNbrPorts;
                 /* Internal hubs have zero bHubContrCurrent.
```

root@raspberrypi:/usr/local/src/uhubctl.new# ./uhubctl 
OCPM: 0
LPSM: 0
OCPM: 0
LPSM: 0
OCPM: 0
LPSM: 0
OCPM: 8
LPSM: 1
OCPM: 8
LPSM: 0
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0507 highspeed power suspend enable connect
   Port 3: 0100 power
   Port 4: 0100 power
   Port 5: 0100 power

mvp, 2017-03-18:
Can you please try building from this branch? https://github.com/mvp/uhubctl/tree/ocpm-relaxed

xxedgexx, 2017-03-18:
root@raspberrypi:/usr/local/src/uhubctl# ./uhubctl 
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0507 highspeed power suspend enable connect
   Port 3: 0100 power
   Port 4: 0100 power
   Port 5: 0100 power

xxedgexx, 2017-03-18:
Actually, I did a clone and I don't think it pulled from the branch, but this one is from the branch:

root@raspberrypi:/usr/local/src/uhubctl# ./uhubctl 
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0507 highspeed power suspend enable connect
   Port 3: 0100 power
   Port 4: 0100 power
   Port 5: 0100 power

Same results.

[ 1889.495573] usb 1-1.2: reset high-speed USB device number 5 using dwc_otg
[ 1889.905561] usb 1-1.2.3: reset high-speed USB device number 7 using dwc_otg
[ 1890.105588] usb 1-1.2.2: reset high-speed USB device number 6 using dwc_otg
[ 2013.654148] usb 1-1.2: reset high-speed USB device number 5 using dwc_otg
[ 2014.064131] usb 1-1.2.3: reset high-speed USB device number 7 using dwc_otg
[ 2014.264109] usb 1-1.2.2: reset high-speed USB device number 6 using dwc_otg

mvp, 2017-03-18:
I see. This hub seems to report 0 for lpsm, which means ganged mode.
Ganged mode means any attempt to change power state will change power for all ports.
However, it might be possible that hub configuration is wrong, and lpsm actually is supported.
You can try hacking it to permit `lpsm == HUB_CHAR_COMMON_LPSM`, and see what happens.

xxedgexx, 2017-03-18:
That didn't seem to work.

mvp, 2017-03-18:
Oh well. Looking at how this 10 port hub is made internally, it is composed from 3 daisy-chained hubs, each of type 2109:2812.

But 7 port hub reported by @phhusson, on other hand, is composed of 2 daisy-chained hubs, each of type 2109:2811. It looks like 2109:2811 supports PPPS (per-port power switching), when 2109:2812 doesn't :(

xxedgexx, 2017-03-18:
No problem.  I'm noticing a pattern here.  It feels like anything beyond 7 ports doesn't support PPPS.  Is there a reason for that?

mvp, 2017-03-18:
Not necessarily. On my home Linux server Asus Z87-PLUS motherboard has internal 8-port hub that works just fine:
```
mvp@linux:~$ uhubctl -i
Current status for hub 2-1, vendor 8087:8000, 8 ports
   Port 1: 0100 power
   Port 2: 0100 power
   Port 3: 0100 power
   Port 4: 0100 power
   Port 5: 0100 power
   Port 6: 0100 power
   Port 7: 0100 power
   Port 8: 0100 power
```

Perhaps real pattern here is daisy-chaining too many hubs together.
For example, very well supported D-Link DUB-H7 (silver edition) has 7 ports, and it is single, not daisy-chained, device.

xxedgexx, 2017-03-18:
The D-Link works well for me.  I'll keep waiting to try something with at least 10 ports and a charging port that works with PPPS.

mvp, 2017-03-18:
I clarified which exact model of AmazonBasics hub works in https://github.com/mvp/uhubctl/commit/5d63424c69eabb1af4048531a711e06e7492dedc.

phhusson, 2017-03-18:
I'm afraid that the answer could be not that it depends on the model, but that they changed their product...
Likewise, I bought https://www.amazon.fr/dp/B009YQAW1S/ Belkin F5U701-BLK which doesn't work... (tried to override uhubctl detection by using hubpower instead of uhubctl, and it didn't work either).
And AmazonBasics doesn't even have any product number to check anything.

I feel like I'll have to brute-force before finding a HUB available in France with PPPS....

mvp, 2017-03-18:
Maybe, maybe not. Here's example: original D-Link DUB-H7 (silver, 7 port) works, while D-Link DUB-H4 (silver, 4 port) doesn't. I'll update info about Belkin hub that not all he revisions work.

phhusson, 2017-03-21:
I've bought a brand "Amazon Basics USB3 7-ports hub", and uhubctl still works perfectly!

mvp, 2017-04-20:
@xxedgexx, it seems that mystery behind difference between those hubs, or USB id 2109:2811 vs 2109:2812 is finally explained.
2811 means it is using Via [VL811 chipset](http://www.via-labs.com/product_show.php?id=43), and 2812 it is using Via [VL812 chipset](http://www.via-labs.com/product_show.php?id=41).
Apparently, those chipsets are quite different.

10dfrc, 2017-11-04:
@phhusson thanks for your work!
I'd like a little more confirmation before buying. So do all HU3770V1 models work, with asin either B00E4J7PTM or B00E6GX4BG? Does the new one you bought have a different asin?

mvp, 2017-11-04:
@10dfrc: I believe all HU3770V1 are the same. Worst case, Amazon has best return policy :) Please report success or failure regardless!

LoganLechat, 2018-01-25:
Just buy : HU3770V1 - USB id 2109:2811 - chipset VL811 - B00E4J7PTM
French amazon and PPPS don't work for me the power stay active on the usb ports
thanks

cruig, 2018-01-26:
@LoganLechat have you tried a USB 2.0 port? When it's plugged in a USB 3.0 port, two 2.0 hubs and two 3.0 hubs are recognized by lsusb, but uhubctl seems to only recognize the two 2.0 hubs, which might be the problem.

I plugged the HU3770V1 in a USB 2.0 port and successfully powered off a port once.

nicolaerosia, 2018-02-03:
@cruig plugged in USB 2.0 I can turn off the ports, any workaround for when it's plugged in a USB 3.0 ?

mvp, 2018-02-03:
I don't have this hub, and not sure what the problem here is. Apparently, this hub works as composite USB device for USB2 and USB3 subcomponents. uhubctl detects only USB2 device with id 2109:2811, not USB3 with id 2109:2810.
Maybe this is an issue with libusb? uhubctl is using standard libusb primitives to enumerate all devices...
If I had access to this hub, it should be easy to debug whats going on here...

nicolaerosia, 2018-02-05:
@mvp, can I help somehow offline?
Or if you can only do it online, I can setup an rpi with the hub attached

mvp, 2018-02-05:
Thanks for offer, @nicolaerosia! Remote access to RPi with this hub connected isn't very useful because RPi is USB2 only.
I need have it locally and try connecting to USB2 and/or USB3 upstream ports to understand what's going on.
I guess I'll order it on Amazon and debug.
This will be tracked in issue #68.