mvp/uhubctl #18: successful(?) usage with Rosewill hub

durka, 2017-04-03:
Hi,

I might have a new device you can add to the table:

http://www.rosewill.com/it-products/computer-components/usb-hub/rosewill-rhub-210-usb-2-0-4-port-hub-with-power-adapter.html

lsusb output is: 0409:005a NEC Corp. HighSpeed Hub

I have an intransigent device attached to this hub, which refuses to reset without unplugging and plugging back in (I tried the USBDEVFS_RESET ioctl to no avail). When I cycle the power with uhubctl, the power LED remains on, however I see the expected messages in dmesg and the device does reset.

Here's the dmesg output after running `sudo ./uhubctl -a off -l 2-3 -p 3` and then `sudo ./uhubctl -a on -l 2-3 -p 3`

```
[Apr 3 17:35] usb 2-3.3: USB disconnect, device number 29
[  +0.214874] usb 2-3.3: new high-speed USB device number 30 using xhci_hcd
[  +0.107419] usb 2-3.3: New USB device found, idVendor=1d27, idProduct=0600
[  +0.000004] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 2-3.3: Product: Structure Sensor
[  +0.000001] usb 2-3.3: Manufacturer: Occipital
[  +0.000002] usb 2-3.3: SerialNumber: 27267
```

Note that _nothing_ happened when I issued the `off` command -- the disconnect and reconnect both happened when I turned it back `on`. Is this expected?

So, does this hub support power control? I'm not really sure, but at least uhubctl is going to solve my problem of resetting this particular device! :)

mvp, 2017-04-03:
You can check if this hub is supported as follows:

- `uhubctl` without parameters should list status for your hub (if your hub is not listed, it is not supported, period).
- Plug some usb device, say flash drive into your USB hub, and confirm that it automounts and you can see flash drive contents ok.
- Try to turn off power for this port using `uhubctl -l 2-3 -a 0 -p 3`. You may need to add `-r N` (where N is some number, 10 to 100) - this may be necessary for some hubs and/or plugged devices to actually turn off power, because they try to turn it back on almost instantly, within milliseconds.
- Confirm that you can NOT access filesystem on your flash drive, so it is effectively turned off.
- `uhubctl` without parameters should now display status for this port as `off`.
- Turn on power using`uhubctl -l 2-3 -a 1 -p 3`. That should make your Linux to reenumerate and mount your flash drive, so it becomes available once again.

If you can follow steps above, then we can safely say that this Rosewill hub is supported.

durka, 2017-04-03:
I confirmed those steps with the same device I was using above; after `uhubctl -a off` I am not able to contact the camera, and after `uhubctl -a on` it works again.

mvp, 2017-04-03:
Good! I will add this hub to supported list then. It would be interesting to know if using `-r N` made any difference.

mvp, 2017-04-03:
Added as supported devuce in 37057c241e3282fca8e6a3ba9fe6c43e91e06fc7

closed 2017-04-03 by mvp

durka, 2017-04-03:
FWIW I don't see any different with `-a off -r 10` or `-a off -r 100`, the power LED still does not go out.

durka, 2017-04-03:
OK, I tried it with a keyboard (looking at the Num Lock light). With `-a off`, light comes back on in less than a second, with `-a off -r 10` same, with `-a off -r 100` it stays off.

durka, 2017-04-03:
The camera is a high speed device. Perhaps that is a relevant difference?

mvp, 2017-04-04:
Thanks for reporting - this is the reason why I implemented `-r` :)
Yes, high speed devices are more hungry for getting power than keyboards, and depending on hub circuitry, they way force hub to automatically turn power back on I guess. `-r` repeatedly sends power off command, which should force hub to stay off.

As for power LED, it really depends on a hub and how well it is designed. For example, for well supported D-Link DUB-H7, each port has own LED, and it does go off when port is turned off by uhubctl.

acjay, 2020-07-04:
I'm not seeing any success with the USB 3.0 version of the hub. Although it advertises per-power switching and appears to be turning off voltage, a USB light I have plugged in isn't going off. I tried with `-r 1000`. Below is the output after running `uhubctl`

```
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0610 4-port hub
  bcdDevice           92.26
  iManufacturer           1 GenesysLogic
  iProduct                2 USB2.0 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Single TT
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      2 TT per port
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0103 power enable connect
   Port 2: 0000.0000
   Port 3: 0000.0000
   Port 4: 0000.0000
```

With `uhubctl -a on` I get 

```
 Hub Port Status:
   Port 1: 0000.0103 power enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
```

I'll make a PR to add this note to the README. I bought this hub specifically thinking it would work with `uhubctl` :(

mvp, 2020-07-04:
@acjay, USB 3 version would be completely different device, and it is not listed currently - so there is nothing to remove from the table.

acjay, 2020-07-04:
I see that now, but the thing is, the model number is exactly the same. I guess this is just my fault, but I didn't notice or think to pay much mind to the USB version column, under the assumption that a major change in the device would lead to a different model number. My point is that in a case like this, making it a little more clear which variants within the model are known to work vs not work could help other folks.

mvp, 2020-07-07:
@acjay, model number for Rosewill USB 3 hub is RHB-210, which is NOT exactly the same as RHUB-210 model of listed USB 2 hub. Please check
https://amazon.com/dp/B00552PMN8

acjay, 2020-07-07:
It says RHUB-210 in the item description on that page, despite the fact that the option selector does indeed say RHB-210. Searching for either term calls up results with both variants, probably due to similar mislabeling and aggregation.

I just think a tiny note similar to the D-Link DUB-H4 would be helpful. I'd be happy to suggest an edit as a PR, but it sounds like you don't agree that this would be helpful.

mvp, 2020-07-07:
@acjay, its weird that description on Amazon is indeed talking about old USB 2 hub, not newer USB 3 one, despite new one having blue USB 3 ports on picture - apparently Amazon copy/pasted it from old hub. I will add a note to the table, thanks for pointing that out.