mvp/uhubctl #23: connect-debounce failed

durka, 2017-04-07:
I have this camera device that doesn't shut down correctly due to poor drivers, so I've added a step to my script that uses uhubctl to power cycle the device before trying to connect to it. It basically looks like this:

```bash
sudo uhubctl -a cycle -d 0.5 -l 2-3 -p 3
sleep 1
connect_to_camera
```

Today after running this several times in succession (trying to fix an unrelated bug...) I saw this in `dmesg` and the camera disappeared from `lsusb`:

```dmesg
[  346.825953] usb 2-3.3: USB disconnect, device number 10
[  347.039389] usb 2-3.3: new high-speed USB device number 11 using xhci_hcd
[  347.146838] usb 2-3.3: New USB device found, idVendor=1d27, idProduct=0600                                                                                             [  347.146841] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  347.146843] usb 2-3.3: Product: Structure Sensor
[  347.146845] usb 2-3.3: Manufacturer: Occipital
[  347.146846] usb 2-3.3: SerialNumber: 27267
[  382.921304] usb 2-3.3: USB disconnect, device number 11
[  383.139235] usb 2-3.3: new high-speed USB device number 12 using xhci_hcd
[  383.246682] usb 2-3.3: New USB device found, idVendor=1d27, idProduct=0600
[  383.246686] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  383.246688] usb 2-3.3: Product: Structure Sensor
[  383.246689] usb 2-3.3: Manufacturer: Occipital
[  383.246691] usb 2-3.3: SerialNumber: 27267
[  400.072990] usb 2-3.3: USB disconnect, device number 12
[  400.287152] usb 2-3.3: new high-speed USB device number 13 using xhci_hcd
[  400.394373] usb 2-3.3: New USB device found, idVendor=1d27, idProduct=0600
[  400.394376] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  400.394378] usb 2-3.3: Product: Structure Sensor
[  400.394380] usb 2-3.3: Manufacturer: Occipital
[  400.394381] usb 2-3.3: SerialNumber: 27267
[ 2059.986028] perf interrupt took too long (2520 > 2500), lowering kernel.perf_event_max_sample_rate to 50000
[ 2149.033317] usb 2-3.3: USB disconnect, device number 13
[ 2161.583398] hub_port_connect: 63 callbacks suppressed
[ 2161.583402] usb 2-3-port3: connect-debounce failed
```

Have you seen this before? Could it be related to uhubctl?

mvp, 2017-04-07:
Side note: `-d` only supports integer seconds delay, perhaps we should fix that.
Try using `-r N` where N is number 10 to 100 - it sends power off many times in a row.
As for real reason: I think `uhubctl` turning power off can confuse Linux kernel about state of USB port. Kernel might be trying to remedy the situation by turning power back again.

Perhaps we should make `uhubctl` more gentle by having it write USB address like `2-3.3` to `/sys/bus/usb/drivers/usb/unbind` to unbind device attached to port of interest before actually turning it off, but this would require root permissions, and would be Linux only. Or, you can do it yourself in your script :)

durka, 2017-04-07:
Additional info:

This is what the USB hub status looks like after the power cycle.

```
Current status for hub 2-3, vendor 0409:005a, 4 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0100 power
   Port 3: 0100 power
   Port 4: 0103 power enable connect
```

But this is not normal. Before doing any power cycles, the port in question was in "highspeed" mode:

```
Current status for hub 2-3, vendor 0409:005a, 4 ports
   Port 1: 0503 highspeed power enable connect
   Port 2: 0100 power
   Port 3: 0503 highspeed power enable connect
   Port 4: 0103 power enable connect
```

durka, 2017-04-07:
I'll try the unbinding before cycle and see if that helps.

mvp, 2017-04-10:
Unbinding seems to make things worse for me - device goes away, but I couldn't find a way to get it back without rebooting.

Perhaps best option is to use `-r N`, where N is 100 or more. Or, change `uhubctl` to keep sending power off many times until device stops automatically powering back on. Not sure how to implement timing logic though - need to experiment on real device with such behavior, but I don't have it available at the moment.

mvp, 2017-04-11:
Well, it seems that not much that can be done from uhubctl point of view here - it does turn power off and on as requested. Please reopen if you think anything can be done by uhubctl to fix this.

closed 2017-04-11 by mvp