mvp/uhubctl #24: Known non-working hubs?

shane-kerr, 2017-04-11:
Does it make any sense to also include USB hubs known _not_ to work? (I realize that such a list might quickly become longer than the working list, but it could be helpful for people thinking of taking a risk for a hub that might work...)

mvp, 2017-04-11:
I don't think so :) As it stands, perhaps 95% (if not more) of USB hubs on market do not support PPPS.

That said, best test is to simply hook it up and see if uhubctl detects it. If it does, it will work 90% of the time. Then report it and we will add it to the list :)

mvp, 2017-04-11:
I was considering adding purchase link for modern devices. But I don't want to promote any particular seller.

However, it perhaps makes sense to add some info on just how readily available particular hardware is. For example, D-Link DUB-H7 silver edition is no longer produced since about 2013, and can be only found on eBay. On other hand, some devices are very modern and are easy to find and buy. I will consider adding a column whether device is currently produced or only has historical value.

shane-kerr, 2017-04-11:
I asked because I have a detected USB hub which does not work (although it is still kind of handy for detecting connection status and so on).

One possibility may be to add a column of the release date of each device (if known), and possibly an EOL for the device too. I'm happy to do a little Internet research to fill this in if it makes sense.

mvp, 2017-04-12:
Added release and EOL year for all supported hubs in 30b17f1894.

closed 2017-04-12 by mvp