mvp/uhubctl #25: Re-enable highspeed?

durka, 2017-04-14:
Some devices on the hub have status "0100 power", but the one I'm power cycling has status "0503 highspeed power enable connect" before cycling, and uhubctl seems to only reset it to "0101 power connect". Usually it immediately returns to 0503, but sometimes it somehow drops off to 0100 instead (and then fails to talk to the computer). Should uhubctl be explicitly resetting the highspeed setting?

mvp, 2017-04-14:
I don't think so. All that uhubctl does is to ask USB hub to turn power off or on for a given port. It should be responsibility of kernel to perform proper enumeration of the device when it reappears on USB bus again.

This is not to say that I'm not open to ideas as what is exactly happening here...

durka, 2017-04-14:
I don't know. I think this hub may not be as well supported as we thought.

For my monitoring I'm now running `./uhubctl -l 2-3 -p 3` in a loop, and I
see the following sequence of events (after the camera has been working
fine for a matter of minutes):

1. Camera software hangs trying to read a frame from the device
2. uhubctl fails to contact the hub at all
3. uhubctl succeeds the next time, but the port status is now 0100,
occasionally 0101, instead of 0503. Also, uhubctl is really slow (~3
seconds), whereas it completed in under 0.3 seconds before #1.
4. It is impossible to bring back the camera, even by replugging the camera
itself. The only thing that works is replugging a *different* device on the
hub.

Totally bonkers.

mvp, 2017-04-14:
What is the model of your hub? Perhaps you can try different hub from supported list?
I had pretty good luck with D-Link DUB-H7 silver edition, but these days it is really hard to find, even on eBay.

durka, 2017-04-14:
This is the Rosewill hub still.

durka, 2017-04-17:
Do you mean [this one](https://www.amazon.com/gp/offer-listing/B00DZS7DLS/ref=dp_olp_0?ie=UTF8&condition=all&qid=1492461178&sr=8-3-fkmr0)? I am skeptical given the prices offered there...

mvp, 2017-04-17:
Old DUB-H7 silver edition works well for me. However, be advised that picture most likely is misleading.
If you order DUB-H7 silver (older) version, you most likely will get black (new version) http://amazon.com/dp/B0000B0DL7, which does NOT support PPPS.

Regarding solving your issue for any hub, I am thinking we might be able to alleviate a problem by calling `libusb_reset_device()` after turning power back on - according to libusb documentation, this API is supposed to reinitiate full USB enumeration for device in question. I will see if this is something I can code in next 1-2 days. In meantime, patches from you are welcome! :)

durka, 2017-04-17:
OK, maybe I will try to order one from eBay where I can contact the seller
and make sure it's the old silver edition.

durka, 2017-04-17:
I'm working on a patch. Can confirm libusb_reset_device causes tons of
dmesg output as the devices reassociate, but as this is a heisenbug I can't
tell whether it's fixed yet.

mvp, 2017-05-10:
Did you have a chance to make use of `libusb_reset_device`? If you have something usable, can you submit pull request? If not, perhaps we should close this issue.

durka, 2017-05-11:
Actually, I did experiment with adding libusb_reset_device. I'm not sure
whether I put it in the right place. In my branch, it resets the entire hub
so all devices reassociate. I'm not sure if there is another place to put
so that only the affected device resets.

Unfortunately I haven't been able to confirm whether it helps or not
because the problem is so inconsistent. It certainly doesn't fix it 100% of
the time.

That said I'd be glad to wrap it up with a command line option and submit a
PR.

mvp, 2017-05-12:
In this case, let's close this issue. Feel free to reopen when you are ready to submit PR.

closed 2017-05-12 by mvp