mvp/uhubctl #26: Just bought 3x AmazonBasics HU3770V1, does not work?

jensolsson, 2017-04-19:
I can turn on and off using the command line, but it does not disable the ports.

Could it be a new revision without the power transistors?

Jens

```
./uhubctl -a 0
Current status for hub 20-1, vendor 2109:2811, 4 ports
   Port 1: 0000 off
   Port 2: 0000 off
   Port 3: 0000 off
   Port 4: 0000 off
Sent power off request
New status for hub 20-1, vendor 2109:2811, 4 ports
   Port 1: 0000 off
   Port 2: 0000 off
   Port 3: 0000 off
   Port 4: 0000 off
```

jensolsson, 2017-04-19:
Thought I'd take it apart and send some photos. Don't know which component to look for.... But i'd bet I am looking for something bigger...

[image]
[image]

jensolsson, 2017-04-19:
Oh this is very interesting

If I do the following:
- Connect one iPhone and one Android Phone to the hub
- Execute ./uhubctl -a 0 -p 1,2,3,4 -l 20-3 && ./uhubctl -a 0 -p 1,2,4 -l 20-1 to turn off all ports
- Nothing happens, both phones are still charging
- I have also verified that they are really charging, increasing battery %

If I instead do the following:
- Make sure no phones are connected to the hub
- Execute ./uhubctl -a 0 -p 1,2,3,4 -l 20-3 && ./uhubctl -a 0 -p 1,2,4 -l 20-1 to turn off all ports
- I attach the iPhone and Android Phone to the hub
- iPhone is NOT charging YES! :), Android is charging NO:(
- Execute ./uhubctl -a 1 -p 1,2,3,4 -l 20-3 && ./uhubctl -a 1 -p 1,2,4 -l 20-1 to turn on all ports
- iPhone are now charging YES! :), Android is still charging
- Both phones show a connection dialog on my computer asking.

So commands seem to have some effect, but I believe there is something in the USB standard telling the iPhone to stop charging and that the voltage is never actually cut.

mvp, 2017-04-19:
Hmm. Apparently, there are different revisions floating around - there are at least 2 other reports that this hub works...

Can you try upgrading firmware? You will need Windows for that.

Does port status change when you try to turn ports ON or OFF? And what happens to plugged USB device like flash drive? Does it go away from host computer point of view?

jensolsson, 2017-04-19:
Hi

I would guess it is software since the date stamp on the PCB seem to match the release date you mention on github so it makes sense it is software fixable then.

Upgrade firmware or even downgrade firmware seem like a good thing to try. Do you have access to older firmwares?

I tried a USB thumb drive now also. The on / off command does not have any effect on it. It is always available and works even if the port is set to off.

Port status change if you mena the status I get when just executing uhubctl without parameters.

Kind regards
Jens

mvp, 2017-04-19:
Unfortunately, I don't have this hub. It was added to the list because 2 other people reported it as working. Devices that worked for me, and I tested them myself:  D-Link DUB-H7, Apple thunderbolt monitor, Apple white USB keyboard, Asus Z87-PLUS motherboard.

jensolsson, 2017-04-19:
>Can you try upgrading firmware? You will need Windows for that.

I cannot find firmware anywhere, do you have a link?

mvp, 2017-04-19:
I have no idea if this will work, but looks like you need to use VL811 chipset updates from http://plugable.com/drivers/hubfirmware/.
Apparently, AmazonBasics is actually made by Plugable, but don't blame me if this doesn't work :)

mvp, 2017-04-19:
Or, get it directly from Via: http://www.via-labs.com/product_show.php?id=43

jensolsson, 2017-04-20:
Trying the via-labs fw upgrade utility does not work for me. It does not find the device it says. Also tried running as Administrator...

Think I am going to try the plugable one, but there are many to select among...

jensolsson, 2017-04-20:
Plugable have the same problem unfortunately...

mvp, 2017-04-20:
Did you upgrade Plugable hub firmware? They seem to have better support for fw upgrades.

I am still confused why some people report them as working perfectly, and some like you don't...

jensolsson, 2017-04-20:
I tried the plugable firmware, but it coult not find the hub :(
Maybe it checks some special signature...

Jens
--
Jens Olsson

jraby, 2017-04-21:
@jensolsson out of curiosity, can you post the kernel logs ( `dmesg`) when the ports are turned off/on ?

jensolsson, 2017-04-22:
059512.622102 USB3.0 Hub             @14900000: IOUSBHostDevice::start: capability descriptors are missing
059512.727181 USB3.0 Hub             @14900000: AppleUSBDevice::start: failed to open _deviceClient
059512.727190 AppleUSBLegacyRoot@: AppleUSBLegacyRoot::usbServiceCallGated: failed to attach or start legacy service
USB3.0 Hub             : bad busy count (0,-1)

Backtrace 0xffffff80008971db 0xffffff8000898b5e 0xffffff800029a8f7 0x0 0x0 0x0 0x0
059525.670610 IOUSBHostDevice@14900000: IOUSBHostDevice::start: capability descriptors are missing
059525.775436 IOUSBHostDevice@14900000: AppleUSBDevice::start: failed to open _deviceClient
059525.775445 AppleUSBLegacyRoot@: AppleUSBLegacyRoot::usbServiceCallGated: failed to attach or start legacy service
IOUSBHostDevice: bad busy count (0,-1)

Backtrace 0xffffff80008971db 0xffffff8000898b5e 0xffffff800029a8f7 0x0 0x0 0x0 0x0

mvp, 2017-05-10:
This issue was open for too long without any activity. Please reopen if you think there is anything actionable we can do on uhubctl to fix this. For now, I consider that there are multiple versions of this AmazonBasics hub - some work, and some do not :(

closed 2017-05-10 by mvp