mvp/uhubctl #27: make says libusb_get_port_path is deprecated

hudsantos, 2017-04-20:
Hello,

I've compiled in a Linux Mint 17, it generated binary and worked fine, but:
```
# make
cc -g -O0 -Wall -Wextra uhubctl.c -o uhubctl -Wl,-z,relro -lusb-1.0
uhubctl.c: In function ‘usb_find_hubs’:
uhubctl.c:345:17: warning: ‘libusb_get_port_path’ is deprecated (declared at /usr/include/libusb-1.0/libusb.h:1358): Use libusb_get_port_numbers instead [-Wdeprecated-declarations]
                 int pcount = libusb_get_port_path(NULL, dev, port_numbers, MAX_HUB_CHAIN);
```

hudsantos, 2017-04-20:
worked fine I mean it has compiled and executed... but it seems the hardware where I tried to run it is not compatible:

**Intel Corporation 82801H (ICH8 Family) USB UHCI Controller**

mvp, 2017-04-20:
Can you​ check what is the value for `LIBUSB_API_VERSION` on your system? Perhaps grep for it under `/use/include/libusb-1.0`. There is ifdef that should use new way to get that information, but it depends on a value of `LIBUSB_API_VERSION`.

As for supporting particular hardware, it's tough call. Very few USB hubs support PPPS (per port power switching) properly, and I am trying to keep compatibility list current. If you find hardware that works, please report!

hudsantos, 2017-04-20:
Hi,

Thanks for replying. Sure I'll let you know if I found hardware that works.

```
$ grep -Ri API_VERSION /usr/include/libusb-1.0/
/usr/include/libusb-1.0/libusb.h: * Since version 1.0.16, \ref LIBUSBX_API_VERSION >= 0x01000102
/usr/include/libusb-1.0/libusb.h: * Since version 1.0.16, \ref LIBUSBX_API_VERSION >= 0x01000102
/usr/include/libusb-1.0/libusb.h: * Since version 1.0.16, \ref LIBUSBX_API_VERSION >= 0x01000102
/usr/include/libusb-1.0/libusb.h: * Since version 1.0.16, \ref LIBUSBX_API_VERSION >= 0x01000102
/usr/include/libusb-1.0/libusb.h: * Since version 1.0.16, \ref LIBUSBX_API_VERSION >= 0x01000102
/usr/include/libusb-1.0/libusb.h: * Since version 1.0.16, \ref LIBUSBX_API_VERSION >= 0x01000102
```

mvp, 2017-04-20:
Hmm. `uhubctl` is looking for `LIBUSB_API_VERSION` [here](https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L338-L346). However, on your system, not only you don't have this macro `LIBUSB_API_VERSION` defined, but even non-compliant `LIBUSBX_API_VERSION is not defined either, just mentioned in comments?

Do you know which version of `libusb` you have installed on your Linux Mint?
libusb 1.0.16 or later must have `LIBUSB_API_VERSION` defined, otherwise it cannot be called `libusb`.

hudsantos, 2017-04-21:
dpkg says I have this version:
ii  libusb-1.0-0:amd64                          2:**1.0.17**-1ubuntu2                                   amd64        userspace USB programming library

I'll take a look. Seems clearly this **is not a issue with your code**.
Even so I appreciate very much your help bro. I'll apt-get update/upgrade/dist-upgrade everything here..
Hope this will solve the problem.

I hope community keeps valuing your work!

closed 2017-04-21 by hudsantos