mvp/uhubctl #28: seems to work but phone still charging

koem, 2017-04-22:
I have a Thinkpad x230 and did this:

`root@lenny:~/code/uhubctl:master ./uhubctl -a off -i`
`Warning: changing port state for multiple hubs at once.`
`Use -l to limit operation to one hub!`
`Current status for hub 2-1, vendor 8087:0024, 8 ports`
`   Port 1: 0100 power`
`   Port 2: 0100 power`
`   Port 3: 0100 power`
`   Port 4: 0100 power`
`   Port 5: 0100 power`
`   Port 6: 0100 power`
`   Port 7: 0100 power`
`   Port 8: 0100 power`
`Sent power off request`
`New status for hub 2-1, vendor 8087:0024, 8 ports`
`   Port 1: 0000 off`
`   Port 2: 0000 off`
`   Port 3: 0000 off`
`   Port 4: 0000 off`
`   Port 5: 0000 off`
`   Port 6: 0000 off`
`   Port 7: 0000 off`
`   Port 8: 0000 off`
`Current status for hub 1-1, vendor 8087:0024, 6 ports`
`   Port 1: 0100 power`
`   Port 2: 0100 power`
`   Port 3: 0103 power enable connect`
`   Port 4: 0100 power`
`   Port 5: 0100 power`
`   Port 6: 0100 power`
`Sent power off request`
`New status for hub 1-1, vendor 8087:0024, 6 ports`
`   Port 1: 0000 off`
`   Port 2: 0000 off`
`   Port 3: 0000 off`
`   Port 4: 0000 off`
`   Port 5: 0000 off`
`   Port 6: 0000 off`

and again which shows the first command powered them all off

`root@lenny:~/code/uhubctl:master ./uhubctl -a off -i`
`Warning: changing port state for multiple hubs at once.`
`Use -l to limit operation to one hub!`
`Current status for hub 2-1, vendor 8087:0024, 8 ports`
`   Port 1: 0000 off`
`   Port 2: 0000 off`
`   Port 3: 0000 off`
`   Port 4: 0000 off`
`   Port 5: 0000 off`
`   Port 6: 0000 off`
`   Port 7: 0000 off`
`   Port 8: 0000 off`
`Sent power off request`
`New status for hub 2-1, vendor 8087:0024, 8 ports`
`   Port 1: 0000 off`
`   Port 2: 0000 off`
`   Port 3: 0000 off`
`   Port 4: 0000 off`
`   Port 5: 0000 off`
`   Port 6: 0000 off`
`   Port 7: 0000 off`
`   Port 8: 0000 off`
`Current status for hub 1-1, vendor 8087:0024, 6 ports`
`   Port 1: 0000 off`
`   Port 2: 0000 off`
`   Port 3: 0000 off`
`   Port 4: 0000 off`
`   Port 5: 0000 off`
`   Port 6: 0000 off`
`Sent power off request`
`New status for hub 1-1, vendor 8087:0024, 6 ports`
`   Port 1: 0000 off`
`   Port 2: 0000 off`
`   Port 3: 0000 off`
`   Port 4: 0000 off`
`   Port 5: 0000 off`
`   Port 6: 0000 off`

Then I plugged in my phone on one of the usb ports and it is still charging - the port is still powered...

mvp, 2017-04-22:
Are you sure you plugged phone in one of affected ports? Some computers have multiple USB hubs. It's possible that you plugged phone into different internal hub, not the ones listed by uhubctl.

koem, 2017-04-23:
Hmm, I wonder how to find out if that's the case ...

mvp, 2017-04-24:
Run uhubctl, check status for all ports. Then connect your device, run uhubctl again, check which ports change their status. This way, you can figure out mapping between physical ports and hub names and ports displayed by uhubctl.

Note that hub id assigned by uhubctl is exactly the same as used by Linux: bus-port1.port2..., but it will work exactly the same on any OS.

If you're on Linux, you can check hub ids in your Linux syslog as they are detected by kernel device enumeration.

koem, 2017-04-30:
ok, I found out my phone was plugged to hub 1-1 Port 2. Still charging after -a off. Seems like it's not possible to switch the power off on the ports

mvp, 2017-05-04:
I guess not much that we can do on uhubctl side - your hardware doesn't seem to support power off properly.

closed 2017-05-04 by mvp