mvp/uhubctl #29: noname hub from aliexpress, works fine

ssinyagin, 2017-04-25:
I couldn't find your email address, so writing here.
This one works:
https://www.aliexpress.com/item/Universal-7-Port-USB-3-0-Hub-With-On-Off-Switch-EU-US-UK-AC-Power/32627558537.html

More details: https://plus.google.com/+StanislavSinyagin/posts/4fFRBVZppuQ

mvp, 2017-04-25:
According to thread you linked, it doesn't really support cutting the power to the port, right?
I wonder if we should add this at all then...

ssinyagin, 2017-04-25:
well, right, I discovered that after I wrote here. uhubctl can cut off the data lane, but not the power.
If you feel we can't do anything, feel free to close this issue. Otherwise I can provide you remote access.

mvp, 2017-04-25:
We have similarly conflicting reports regarding AmazonBasics hub...
I know for sure that D-Link DUB-H7 (silver) does cut the power.
Maybe this is manufacturer going cheap on implementing switching circuitry...
I'll add a note about this in README - that some hubs cut the data lines, but not power lines.

closed 2017-04-25 by mvp