mvp/uhubctl #35: Raspberry Pi 3?

jensolsson, 2017-05-24:
Anyone know if raspberry pi is supported? any version? 3? Figured since it has 4 ports it is almost like a usb hub?

mvp, 2017-05-24:
Not to my knowledge, but I have not tried myself.
All we need is for someone who owns Pi 3 to compile and try it.

SlySven, 2017-05-24:
Well - looking at the schematics [B](http://www.raspberrypi.org/wp-content/uploads/2012/04/Raspberry-Pi-Schematics-R1.0.pdf) [2B](http://www.raspberrypi.org/wp-content/uploads/2012/10/Raspberry-Pi-R2.0-Schematics-Issue2.2_027.pdf) that are out there for the RPiB+/2B I can tell you that the LAN9512/9514 Chips those use for 2/4 USB (+Ethernet) hubs have pins marked as PRTCTRL# where # is a digit - but they are floating and unused.  TBH given the limited USB power capabilities of the on-board 3.3V supply - especially for the earlier generation units - the absence of power control for the USB ports is an annoying deficit.  OTOH The Foundation probably omitted this on grounds of additional cost and possibly PWB real-estate.

Whilst the RPi3 has some differences I do not believe that this is one of them. 😢

mvp, 2017-06-01:
I do not think that here is anything to do on uhubctl side.
If Raspberry Pi would support PPPS, uhubctl would work.
But now, this is not actionable - closing.

P33M, 2017-06-12:
Hi. I work for Raspberry Pi and was linked here from a community forum thread: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=185898

The following Raspberry Pi models support (ganged) power switching:
- Model B+
- Model 2 B
- Model 3
But there is a caveat for how it is implemented. The integrated Ethernet/USB Hub controller defaults to per-port power switching in its hub descriptor, whereas the electrical connection arrangement for the overcurrent sense chip is for ganged power switching.

Using uhubctl on Port 2 of the integrated hub will toggle power to all downstream physical ports simultaneously.

SlySven, 2017-06-12:
Does that mean that using `uhubctl` on Port **1** could disable the wired-ethernet - users had better be careful if accessing the RPi over an SSH hard-wired network connection then, otherwise they could accidentally cut themselves off! 😮

P33M, 2017-06-12:
No, it doesn't. The Ethernet controller is connected to port 1 and is non-removable, so a) the power is not removed from the device and b) the kernel doesn't assume the power has been removed from the device.

mvp, 2017-06-12:
I don't have Raspberry Pi to test. @P33M or @SlySven, can you try compiling and running uhubctl on RPi from this branch: https://github.com/mvp/uhubctl/tree/ocpm-relaxed? Try if it works on port 2 of your integrated hub. You should use something like
uhubctl -i -p 2.
Thanks!

mvp, 2017-06-13:
If this fix makes it work on RPi, I would be very happy to add RPi to compatibility list. But I need someone to test it on various RPi hardware editions to be sure.

P33M, 2017-06-13:
It works. The -i option is superfluous, as the hub is not part of the root hub. Setting port power on port 2 has the desired effect on external downstream ports. Setting port power on port 1 has no effect.

mvp, 2017-06-13:
Support for Raspberry Pi is committed in c3da09da55e1919c875b45b9aa05bf38593ffbcd.

closed 2017-06-13 by mvp

SlySven, 2017-06-15:
@P33M wrote:
>No, it doesn't. The Ethernet controller is connected to port 1 and is non-removable, so a) the power is not removed from the device and b) the kernel doesn't assume the power has been removed from the device.

On my RPI 2 B using `uhubctl -p 1 -a 0` DOES switch off the wired LAN port without notifying the kernel or networking driver software - which is not nice if you are talking to your RPi over a `ssh` session via the wired network connection! 😱

Typing/using `uhubctl -p 1 -a 1` _locally_ afterwards does not seem to be enough to restore operation but `uhubctl -p 1 -a 1 -R` does do the job... 😌 

Oh, and `uhubctl -p 2 -a 0` and `uhubctl -p 2 -a 1` does work to switch __all__ the USB ports on and off simultaneously but not on RPi B models before the 1+ (on my other RPi a  1.2B, with 2 USB ports which predates the 1+ B, the hub IC does respond but I think the power control lines do not appear to be wired up and so it does not control anything).

P33M, 2017-06-16:
The interaction is a bit more subtle. I suspect it's a result of the Ethernet controller being connected via back-to-back UTMI or HSIC.

If I disable "power" to the port, no messages are generated in dmesg. Pings to hosts fail with a timeout. If I subsequently re-enable the port power, the Ethernet controller disconnects and then immediately reconnects. Pings start working again for me.