mvp/uhubctl #37: Need pics of working ppps hubs

k1tzu, 2017-06-06:
Can somebody please send high-res pics of both sides internals of Anker AK-A7518111 or UHR204 or Plugable USB2-HUB10S or any that has really working ppps?
That will be of much help.
I already have pics of Amazon hub, need more to compare.
Thank you a lot!

mvp, 2017-06-06:
UHR204 is quite expensive industrial grade hub, I doubt anyone would find pictures for you.
You might be better off just buying Anker hub from Amazon for $15.

k1tzu, 2017-06-06:
I live in Russia, I can't order it even from Amazon. It will cost like 40$ to send it here.

mvp, 2017-06-06:
You may have some luck by simply trying all USB hubs you have access for, just try to connect them to your Linux and see if uhubctl detects it, some might just work. This is what I did myself :)

mvp, 2017-06-06:
Microchip EVB-USB2517 eval board is supported, it has quite comprehensive schematics here http://ww1.microchip.com/downloads/en/DeviceDoc/evb2517sch.pdf.

k1tzu, 2017-06-07:
Thank you. Yes I've bought several hubs, last one was DLINK DUB-H7 ver C2. None of them were working with ppps.

mvp, 2017-06-07:
D-Link DUB-H7 works great, but only old silver edition (A series). New black (C series) doesn't work.

mvp, 2017-06-13:
Closing this due to inactivity, and it's not actionable from uhubctl development standpoint.

closed 2017-06-13 by mvp