mvp/uhubctl #38: Zero Pi Support?

AlessandroEmm, 2017-06-13:
Hey there 

Is there a limitation in Zero Pi's Support for what the bigger Pi's support?

Thanks a lot!
Alessandro

mvp, 2017-06-13:
Raspberry Pi compatibility list was provided by Raspberry Pi employee [@P33M](https://github.com/P33M) in #35.
I guess Zero Pi does not have proper hw.

SlySven, 2017-06-15:
The RPi Zero does not have a hub - it only has one proper micro-USB (on-the-go capable IIRC) port (the Power supply input - although also a micro-USB does not have connected data lines I believe) although, of course you could use `uhubctrl` to control a connected hub...!

mvp, 2017-06-15:
Sure, connecting any of supported hubs to Zero Pi would provide ability to control power. The only trouble of course would be that Zero Pi would be smaller in size than almost any USB hub :(
Closing issue as not actionable from uhubctl standpoint.

closed 2017-06-15 by mvp

pssc, 2017-06-15:
https://thepihut.com/products/zero4u-4-port-usb-hub-for-raspberry-pi-zero ?

SlySven, 2017-06-15:
The above item looks nice and compact (though it only has a single transaction unit so using a low speed 12MBs device on one port will slow all the other three down) - if it __DID__ support PPS it would be very nice but I'd be surprised! Perhaps the manufacturers can be made aware of this GitHub project (should it not do power control) so they could consider it for the next version - as of course the RPi Zero is _highly_ likely to be used in a situation where battery powering is involved and power control would be very desirable... 😀

mvp, 2017-06-16:
Yes, it would be very nice if USB hub manufacturers supported PPPS more widely. I think many people would happily pay $5-10 extra for USB hub that supports PPPS, even if adding power switching circuitry would cost extra few cents. As far as I know, virtually all USB chipsets DO support PPPS, but manufacturers typically disable it on chipset level.

xxedgexx, 2018-09-09:
Has anyone tested if this works with a pi zero?

mvp, 2018-09-09:
@xxedgexx, I assume your question isn't about Pi Zero supporting PPPS, but rather about this particular stackable 4 port hub supporting PPPS. I don't believe anyone tried it. If you do, please report back if it works with uhubctl - thanks!

xxedgexx, 2018-09-09:
@mvp I'm going to get this thing and see if it works with PPPS and the pi zero w.  Thanks!

https://www.amazon.com/MakerSpot-Accessories-Charging-Extension-Raspberry/dp/B01JL837X8/ref=sr_1_2?ie=UTF8&qid=1536475514&sr=8-2&keywords=pi+zero+usb+hub

xxedgexx, 2018-09-10:
This:

https://www.amazon.com/MakerSpot-Accessories-Charging-Extension-Raspberry/dp/B01JL837X8/ref=sr_1_2?ie=UTF8&qid=1536475514&sr=8-2&keywords=pi+zero+usb+hub

Does not work.  No compatible hub is recognized.

hichiaty, 2022-04-21:
has anyone tested if this supports PPPS? https://thepihut.com/products/zero4u-4-port-usb-hub-for-raspberry-pi-zero