mvp/uhubctl #39: Working Hub: Maplin Product Code A08CQ (No longer stocked)

SlySven, 2017-06-20:
Was just testing uhubctl on the hubs I do have and found this fairly decent one I picked up at a car-boot sale (Vendor:Product = `0409:0059`) seems to work:
[image]

It has a green port status LED for each of the 7 ports (two on front and five + USB B input port on rear) it also has a dual colour "Power" LED which is green if it is run from the upstream and "orange (external power & upstream connected)/yellow (external power only)" and a 5V 3.5A centre positive 2.5mm barrel connection for external power.

From the display from `lsusb -t` it does appear to be two four-port units with one cascaded from the fourth port of the other.  This is a little confusing though! How do I specify that I would want to change the power setting for a single device plugged into the fourth port of the second hub (to which I've plugged in a USB Thumb drive for testing purposes:

    $ sudo ./uhubctl
    Current status for hub 2-4.4, vendor 0409:0059, 4 ports
       Port 1: 0100 power
       Port 2: 0100 power
       Port 3: 0100 power
       Port 4: 0503 highspeed power enable connect
    Current status for hub 2-4, vendor 0409:0059, 4 ports
       Port 1: 0100 power
       Port 2: 0100 power
       Port 3: 0100 power
       Port 4: 0503 highspeed power enable connect

Turning the power off to a port does turn the appropriate Port LED off if there is a device connected (and the device becomes inaccessible) and turning the power back on reverses that - so it does seem to behave as I'd expect...! 😀 
 
Sadly Maplin no longer seem to sell this unit, but you might find one on an second hand auction site or elsewhere...

mvp, 2017-06-21:
Added in 93173f6.

closed 2017-06-21 by mvp

SlySven, 2017-07-06:
Actually seeing @mvp 's response to #40 suggests the answer to my question in the above would be to use:

    uhubctl -l 2-4.4 -p 4 -a #

to refer to the Port 4 on the __chained__ second port attached to Port 4 on the first 4-port hub (where # is one of `0`, `1` or `2`).  To clarify the Hub (or rather the first 4 port part of it) is plugged into port 4 of my PCs second USB root (motherboard) hub i.e. it is at `2-4` and the second part - which is daisy-chained off of the fourth port of the first - is at `2-4.4` and to control something that is plugged into the fourth port on the second hub I need to use the `-p 4` - theoretically (and actually) I can turn off __just the second daisy-chained hub__ with either:

1. `uhubctl -l 2-4.4 -a 0`
2. `uhubctl -l 2-4 -p 4 -a 0`

In 1. the second hub is told to turn off all its ports; in 2. the upstream hub is told to turn off it's output to the second  - which may not revert/reset so well.

In my particular case because the second hub is itself still powered (although all its outputs are turned off) in case 1. it still shows up in the `uhubctl` status listing, whereas if the first port has the output turned off (as per 2.) it also removes the power from the second hub so that no longer appears in the status listing (and probably saves a fraction more power if that is _really_ critical)....

mvp, 2017-07-06:
I would still refrain from ever touching ports used for daisy-chaining - we don't know how exactly it is wired internally by hub manufacturer, and whether cutting power to master port will actually power off slave device.
By the way, last version of uhubctl (1.7) shows attached device identity (vid:pid, product and perhaps serial number), which should make it obvious when daisy-chaining is used - you would see that some hub is connected to the port, when no actual physical devices are connected.

SlySven, 2017-07-07:
Ah, I need to update (only got 1.5)... 😊