mvp/uhubctl #40: uhubctl showing hub twice?

tinnocente, 2017-07-05:
Hi very nice software.

I purchased the Amazon basics 7 port hub as recommended and when I run `./uhubctl` as root the Amazon hub is displayed twice.

> Current status for hub 3-1.5.3, vendor 2109:2811, 4 ports
   Port 1: 0100 power
   Port 2: 0100 power
   Port 3: 0100 power
   Port 4: 0100 power
Current status for hub 3-1.5, vendor 2109:2811, 4 ports
   Port 1: 0103 power enable connect [0801:0013 Mag-Tek USB Insert Reader ?J??:?]
   Port 2: 0100 power
   Port 3: 0507 highspeed power suspend enable connect [2109:2811 VIA Labs, Inc.          USB2.0 Hub             ]
   Port 4: 0100 power

> 

If I try to control ports by turning them off, it works. However when I try to turn them back on or cycle the power, the software hangs after sending the power off command. I am suspecting the problem is the fact that the hub is being detected twice.

I am running Debian Jesse 8.7 (3.16 kernel)

Any ideas?

Thanks!

mvp, 2017-07-05:
This is correct behavior. This hub is actually 2 4-port hubs which are daisy chained internally together - second hub is wired to port 3 of master hub.

tinnocente, 2017-07-05:
Thanks for the quick reply - that makes a lot of sense.

Any suggestions how to cycle power of port 3 of the `3-1.5` and not ` 3-1.5.3`?

mvp, 2017-07-05:
Actually, you don't want to ever touch port 3 of `3-1.5` - this is daisy chain port to second hub. Turning it off should make second hub completely disconnected, which depending on hub circuitry may give all kinds of surprises.

To operate on port 3 of slave hub`3-1.5.3`, simply specify hub location using `-l` option, something like this:

    uhubctl -l 3-1.5.3 -p 3 -a 2

This would reset port 3 of second hub in chain.
For location parameter, you need to use hub id that uhubctl prints - it is basically the same id that you would see in Linux kernel dmesg and other logs when hub is first enumerated.

tinnocente, 2017-07-06:
Thanks I will try it out tomorrow.

Would you know of a hub with 5 or more ports that don't daisy chain? 

Thanks again!

mvp, 2017-07-06:
D-Link DUB-H7 is true 7 port hub with power indicators per each port that works really well. But you need to find old silver edition (perhaps on eBay), as new black edition doesn't support PPPS.

tinnocente, 2017-07-06:
Thanks for confirming!

closed 2017-07-06 by tinnocente