mvp/uhubctl #41: Arch Linux permission problems

severach, 2017-07-12:
Please add column to table USB 2.0, 3.0, or 3.1. This is a major purchase decision so you may want to sort by it.

Reference [Use Linux to control power outlet via USB](http://tiebing.blogspot.de/2011/01/use-linux-to-control-outlet.html)
Reference [USB Per Port Power Switching](https://rwmj.wordpress.com/2015/10/09/usb-per-port-power-switching/)

***

It's not working in [Arch Linux](https://aur.archlinux.org/packages/uhubctl/). I don't have any hubs yet but two motherboard hubs claim to have the feature so seems to me the permissions error shouldn't be shown.

````
$ sudo uhubctl 
No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'
$ sudo lsusb -v 2>/dev/null | grep -e '^Bus\|Per-port power switching' | grep -B1 'Per-port power switching'
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
    Per-port power switching
--
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
    Per-port power switching
$ pacman -Q libusb
libusb 1.0.21-2
````

mvp, 2017-07-12:
You need to add option `-i`, or use explicit `-l <location>` to operate on onboard USB hubs.
But default, onboard hubs are not shown unless `-i` is used.
This is to avoid people accidentally turning off power for keyboard or mouse which are typically connected to onboard hubs.

Regarding adding USB version column - its not that many hubs that support PPPS are out there.
The only USB 3 hub that supports it has USB3 in description, all others are USB2.
 But I agree that over time, we should consider adding this.

closed 2017-07-12 by mvp

severach, 2017-07-12:
`There were permission problems while accessing USB.`

This should not be shown. All it should say when I don't have any supported hubs is `No compatible smart hubs detected!`

mvp, 2017-07-12:
Perhaps simplest solution for this is to make `-i` always active (and remove this option altogether).
I only added `-i` because it was too easy to cut my own keyboard while debugging :)

severach, 2017-07-12:
`-i` works fine. It's that `There were permission problems while accessing USB.` suggests that something is broken. It's a good message when things are broken but it's a bad message when things are not broken. To not have any supported hubs is not broken.

mvp, 2017-08-02:
Fixed in v1.8 by 047bb1e39.