mvp/uhubctl #42: uhubctl hangs when turning device power back on

tinnocente, 2017-07-12:
Seems the software is buggy on my system.
Using the Amazon Basics 7 port hub on Debian Jesse 8.7 (3.16 kernel).
All commands are being run as root user

These commands work as expected:

> # ./uhubctl
> Current status for hub 3-1.5.3, vendor 2109:2811, 4 ports
>    Port 1: 0100 power
>    Port 2: 0100 power
>    Port 3: 0100 power
>    Port 4: 0303 lowspeed power enable connect [04ca:002f Lite-On Technology Corp. USB Multimedia Keyboard]
> Current status for hub 3-1.5, vendor 2109:2811, 4 ports
>    Port 1: 0100 power
>    Port 2: 0103 power enable connect [0c2e:0aaf Honeywell Imaging & Mobility     1910i 17033B01A3]
>    Port 3: 0503 highspeed power enable connect [2109:2811 VIA Labs, Inc.          USB2.0 Hub             ]
>    Port 4: 0303 lowspeed power enable connect [046d:c00c Logitech USB Optical Mouse]
> 
> # ./uhubctl -l 1-1.5.3 -p 3 -a 2
> Current status for hub 1-1.5.3, vendor 2109:2811, 4 ports
>    Port 3: 0100 power
> Sent power off request
> New status for hub 1-1.5.3, vendor 2109:2811, 4 ports
>    Port 3: 0000 off
> Sent power on request
> New status for hub 1-1.5.3, vendor 2109:2811, 4 ports
>    Port 3: 0100 power

This command hangs:

> # ./uhubctl -l 1-1.5 -p 4 -a 2
> Current status for hub 1-1.5, vendor 2109:2811, 4 ports
>    Port 4: 0303 lowspeed power enable connect [046d:c00c Logitech USB Optical Mouse]
> Sent power off request
> New status for hub 1-1.5, vendor 2109:2811, 4 ports

The scary part is when it hangs I cannot kill it with CTRL+C or even with kill -9 PID

Any ideas would be much appreciated - thanks!

mvp, 2017-07-12:
I guess you'll need to debug this - I don't have this hardware in my posession...
You might be able to get it to work by using `-R` option - it attempts to reset usb devices connected to hub in question.
Fact that you cannot kill it suggests that this might be kernel USB hub driver issue.
All uhubctl functionality about turning power on or off is basically done in this call:
https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L601lL605

    rc = libusb_control_transfer(devh,
        LIBUSB_REQUEST_TYPE_CLASS | LIBUSB_RECIPIENT_OTHER,
        request, USB_PORT_FEAT_POWER,
        port, NULL, 0, USB_CTRL_GET_TIMEOUT
    );

`libusb_control_transfer` is eventually calling kernel USB hub driver to get this work done.
Can you try adding some `printf` statements before and after this call?

mvp, 2017-07-25:
Closing for inactivity. Please reopen if you have more details about this freeze.
Would be also interesting if you could try this on different Linux distro like Ubuntu, and/or perhaps different compatible USB hub - it works for me very reliably, I'm using Ubuntu 16.04 and DLink DUB-H7.
Thanks!

closed 2017-07-25 by mvp