mvp/uhubctl #43: Inland Brand USB 3.0 hub with Per-Port Power Switching

aciani, 2017-07-21:
An Inland brand (www.ipsgproducts.com) model UPH341 (Micro Center SKU 902064) supports per-port power switching.  Inland is Micro Center's house brand.

This is a 4-port USB 3.0 bus-powered hub.  It identifies as 2109:2813 VIA Labs, Inc.  Some devices play nice and power off with a single command.  Had a few that needed multiple repeats (3 with 200 ms waits).

This is a US $11 hub.

aciani, 2017-07-21:
Nevermind.  It looks like it does not physically or electronically cut power.  The hub controller IC cutting the serial link, but not cutting power.

closed 2017-07-21 by aciani

mvp, 2018-02-18:
@aciani: Can you please try this one again using uhubctl built from master branch?
Be sure to turn off power for both USB2 and USB3 virtual hubs. Thanks!

aciani, 2018-02-19:
No need.  The Inland hub has a hard-wired power bus.  I popped open the
case and took a look when uhubctl failed to switch the LED snake light I
used.

I have popped open several USB hubs, and have found all to date to have
hard-wired power buses.  I have also found that their radio frequency
harmonic filter circuits are missing, which is why so many of them make
disagreeable squeals when plugged in.  I doubt that most USB 3.0 hubs can
actually function at full USB 3.0 speeds due to self-generated RF
interference (BTW, clearly audible on AM radios).  Many of the hubs I
popped open had spots for the components, but no components.  The
difference between the $9 hub and the $59 hub is the $50 extra you send to
China.  Both are non-compliant garbage with false claims covering their
packaging.

There were a couple hubs that I did not pop open.  I'll give them a try
with the master branch and report any successes.

I brought the issue up with the USB consortium.  They were surprised at how
wide-spread the non-compliance was, but were not terribly surprised that
Chinese assembly houses were stripping all but the most critical components
from the designs or bill of materials.  I'm not sure they are going to do
anything.

Also to note.  I have heard that the Chinese will often manufacture a US
version and an EU version of the exact same part number, because the EU is
much more strict on compliance and watchful for garbage from China.  We
might have better luck with hubs sold in Europe.

mvp, 2018-02-20:
Thank you, @aciani! I didn't know it's even possible to raise non-compliance issues with USB consortium. Can USB consortium revoke right to carry USB logo for non-compliant product? Itould be really nice if manufacturers started to actually pay attention to these things.

SlySven, 2018-02-20:
Might one suggest it is something the retailers should have to start paying attention to, as I suspect if what they are selling is not _doing what it says on the tin_ they will end up carrying the _can_ when purchasers start demonstrating that the products they are selling is incapable of doing what the labels claim and demanding their money back - after all, as a customer, you give your money to the retailer and that is who you have the trading relation with...

Ideally, you'd want to make sure when you buy a Hub that you make it clear in front of an impartial third party that PPPS is something that you expect and require and that it is a proper part of the USB specification (having a summary to hand might not be a bad thing if the retailer is going to pretend to not understand what you are saying/asking for) -  so that when it doesn't work it is clear that they didn't sell you what you explicitly asked for!

Remember I am not a lawyer - but you are a customer and the customer is always right (maybe misguided, confused or mis-informed) but still __right__. 😈

mvp, 2018-02-20:
Unfortunately, I don't believe that customer complaints will significantly improve quality here.

We need USB consortium as fire marshal to raise their voice and remoke USB logo rights from manufacturers who sell non-compliant (and potentially, hazardous) products.

After all, if USB hub manufacturer skips circuitry to actually cut power to the port (to save few cents), it means OC (overcurrent protection) is non-functional, and this becomes real fire hazard. This is when fire marshal should step in and prohibit sale of potentially hazardous product.

aciani, 2018-02-23:
It is possible to contact the individuals on the USB Implementers Forum
(USB-IF) governing board, but most industry consortia are essentially
internally-funded, minor projects for the employees involved, and USB-IF is
no exception.  Compliance is handled in a tiered manner.  USB-IF sets the
standards and testing methods and licenses the USB trademark to the IC
foundries (e.g. Silicon Image) who demonstrate that their IC's, when used
in a reference circuit, perform to the standard.  The IC suppliers in turn
sub-license the USB trademark to manufacturers who fabricate products that
use their USB ICs.  It seems that the IC suppliers leave it to the retail
manufacturers to certify their own products; then again, given the state of
the industry, the prototype that was certified might not be what the board
house manufactures.

The USB-IF (like most consortia) will only take action when it needs to,
which will be when the non-compliance problem is large enough that it
negatively impacts IC foundries or manufacturers who honestly try to
comply.  Unfortunately, technical incompetence among electrical engineers
and customer support is so wide spread that even the most legit companies
are non-compliant and unaware.

For those interested in what can be done, there are two approaches to bad
electronics: market and regulatory.  I suggest the market approach first.
Government tends to be slow or nonreactive.

Market approach:
1) Purchase a bus-powered hub and test for port or ganged power switching
and over-current protection.  All bus-powered hubs must support switching
and protection, per USB standard.  Over-current protection is provided
through current monitoring and power switching.  If the hub cannot switch
or cannot monitor current, then it has no power protection.
2) Contact manufacturer's customer support (CS) and notify them of
non-compliance.  Log all contacts with CS.  For example, Anker competently
answered my inquiry and stated that they would correct the issue in a new
hub model, so no further action required at this time.
3) If CS provides no satisfaction (e.g. "our hubs depend on the upstream
port for power protection" - SIIG, or "please provide a picture" -
Patriot), then write a letter (the dead tree type) to their CEO explaining
the problems (e.g. non-compliance with USB standard, radio interference,
false claims on packaging, plus your nose full of milk or split sides) and
why their CS's response was inadequate.  If you can identify their
corporate legal contact, mail a copy of the letter to them as well (the
letter [not the envelope] should be addressed to both recipients).
Note: Social media is useless.  Those guys just dole out apologies, refunds
and coupons to oil squeaky things.

Regulatory approach:
1) Radio interference is handled in the United States by the Federal
Communications Commission (FCC).  If the use of the USB hub interferes with
radio reception (i.e. you hear it in the radio signal), then it is harmful
and non-compliant.  File a complaint with the FCC.  I recommend checking
for interference in the AM bands, as FM is less susceptible to it.  Provide
everything you can about the hub, which pretty much means every number or
code on the packaging and device, especially the FCC ID.
2) False claims are handled in the United States by the Federal Trade
Commission (FTC).  If the packaging claims power switching or over-current
protection, then it is a false claim.  You will need pictures of the
packaging, a clear description of the testing you performed and a concise
description of why the packaging claim is false.  The FTC is generally slow
to act; however, a wide-spread (industry-wide) problem is more likely to
receive action.  The use of the USB trademark on an item that fails the USB
standard is also a form of false claim.
Note: Many states in the USA have their own laws and enforcement agencies
regarding deceptive labeling.
3) Although small, hubs with hard-wired power buses do pose a fire risk and
risks to upstream equipment (i.e. motherboard).  In the United States, this
could be addressed to the Consumer Products Safety Commission (CPSC).  If
the hub has an Underwriters Laboratories logo (most low-voltage equipment
does not), contact UL.  You might also want to contact your state's Fire
Marshall to find out if your state has any regulation or enforcement.