mvp/uhubctl #46: Supported Hub: CyberPower

Trevor-L, 2017-07-28:
### CyberPower

    Model: CP-H420P
    USB 2.0 4-Port Hub with Power Adapter

```
Current status for hub 20-1, vendor 0409:0059, 4 ports
   Port 1: 0100 power
   Port 2: 0100 power
   Port 3: 0100 power
   Port 4: 0100 power
```

mvp, 2017-07-28:
Thank you! Did you try to actually turn power off and see that power is physically cut? Some hubs report this feature, but don't really cut power.

mvp, 2017-07-28:
Added in 18ea2fe35a. Thank you for reporting!

closed 2017-07-28 by mvp

Trevor-L, 2017-07-31:
I plugged an iphone in and it seems to cut charge power.  However, when using some woot! lights that just pulls power it would not cut power to them.

mvp, 2017-07-31:
Argh. I still don't quite understand how those USB chipsets work.
Lets keep this in the list though - cutting power to changing phone seems to be working at least partially.

aciani, 2017-08-01:
I reviewed the spec sheet for an NEC hub controller (uPD720112), and it had timings for the process of over-current detection.  When the over current detect pin is pulled low, it needs to wait 500 microseconds for the condition to clear (to filter possible in-rush current).  The controller then sends a bus reset and brings the port power pin high (to switch off the FET and cut power).

Basically, the controller first asks the device to shutoff and play nice, then it forces the device off by cutting the +5 V line.  So when the controller is told to shutoff the port, it follows the same process of voluntary shutdown followed by forced shutdown (if the FETs or relays are present).

Most baffling about all of this: the USB specs require power switching on bus-powered hubs.  The controllers support this, but the manufacturers leave out the extra bits required for it to work.  The hubs are then not truly USB compliant.

NaterGator, 2021-01-09:
I have one of these and it is _not_ cutting VBUS for me. Confirmed via AVHzY USB power meter. VBUS doesn't budge, regardless of whether the hub is self-powered or powered from upstream.