mvp/uhubctl #47: Execution error on Debian Stretch

ssinyagin, 2017-07-31:
uhubctl doesn't work on Debian 9.1:

```
root@asrock:/opt/build/uhubctl# ./uhubctl 
No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'
```

ssinyagin, 2017-07-31:
it worked on another Debian 9.1 host though... any ideas? This machine under Debian 8 worked fine with uhubctl

mvp, 2017-07-31:
This is just incorrect error message - use option `-i` to fix. I plan to make -i always on.

ssinyagin, 2017-07-31:
indeed, it worked. Thanks.

closed 2017-07-31 by ssinyagin

mvp, 2017-08-02:
Fixed in version 1.8 by 047bb1e3.