mvp/uhubctl #48: Add to compatibility list: PC Engines APU 2 and 3

ssinyagin, 2017-07-31:
PC Engines APU2 and APU3 boards have a USB controller with power management. APU1 has a different controller, without power management.

http://pcengines.ch/apu2.htm

```
Bus 002 Device 002: ID 0438:7900 Advanced Micro Devices, Inc.
```
APU2 has one such controller, connected to its J13 mini-PCIEe slot suitable for 3G/LTE modems.
APU3 has two such controllers, connected to J14 and J15 mini-PCIE slots (the board supports two 3G/LTE modems)

uhubctl is able to disconnect and connect the modem in those slots.

ssinyagin, 2017-07-31:
arghh never mind, it only disconnects the data lane. The attached modem is not powered off by uhubctl

closed 2017-07-31 by ssinyagin

pietrushnic, 2017-08-04:
@ssinyagin I'm apu1/2/3 firmware maintainer is there anything I can help you in your issues ?

ssinyagin, 2017-08-04:
@pietrushnic it's not really an issue, but inconvenience. And I believe it's not related to firmware, but rather to hardware schematics: 

The USB hub behind the mPCIE interface reports itself as power management capable, but in fact it cannot cut off the +5V power from the socket. I think it's just the way the power is wired to the socket. 

maybe this can be addressed in future versions of APU board, but it's not too critical. It's very rare that an LTE modem needs a hard reset.