mvp/uhubctl #49: Charging Continues After Port Shutoff on Hubs

baclark774, 2017-07-31:
Running this on latest version Mac osX sierra. Tried it on a couple of different supported usb hubs. Most recently the Anker AK-A7518111. I have multiple ios and android devices connected to the usb hub. When I disable charging via "uhubctl -a off" i have to physically disconnect/reconnect a tethered device/usb connection to the hub for the it to stop charging. Data connectivity is terminated, but charging continues unless someone manually disconnects/reconnects the device.

Is this expected? Looking to use this as a cost effective solution to not continuously charge my tethered development devices.

mvp, 2017-08-01:
This is second report of Anker A7518111 not working properly. I guess we should remove it from compatibility list :(

aciani, 2017-08-02:
Definitely.  Anker replied that the A7518111 does not offer power switching.  Waiting to see at what revision this feature was removed, if it ever existed.

aciani, 2017-08-02:
Anker informed me that they do not have any power switching capable hubs in production, but are working on a new design that will.

I checked the USB-IF compliance advisories, and it looks like there was a recent advisory concerning power switching compliance.  Downstream port power switching is required by USB Specification 2.0 section 7.2.1.1 for bus-powered hubs, and section 7.2.1.2 for externally powered hubs that can be bus-powered.  This specification is inherited by USB 3.0 devices.  It seems incidents of fried motherboards from over-current hubs are increasing as more hubs are being manufactured non-compliant.

mvp, 2017-08-02:
Removed this hub from compatibility list in a83b28f2.

closed 2017-08-02 by mvp

mvp, 2018-02-18:
@aciani: Can you please try this one again using uhubctl built from master branch?
Be sure to turn off power for both USB2 and USB3 virtual hubs. Thanks!