mvp/uhubctl #50: Struggling to Find Working USB Hub

baclark774, 2017-08-06:
Hi team,

Really appreciate the effort that went into this. I previously tried the anker model without success and that has been removed. But today I tried both the Amazon Basic HU3770V1, 7 Port USB 3.0 Hub with 12V/3A Power Adapter item, and the Pluggable USB2-HUB10S. The Amazon Basic USB had the same issue as the anker. The pluggable usb hub gets a "No compatible smart hubs detected!" when running from a mac.

Can anyone recommend a confirmed working usb hub? Would really appreciate it as I don't want to have to keep buying and trying them all from the list.

mvp, 2017-08-06:
D-Link DUB-H7 old silver edition definitely works.

mvp, 2017-08-07:
Regarding other reports, like AmazonBasics HU3770V1 or Plugable USB2-HUB10S, those were reported by various users, and they were working for them at a time. It's quite possible that those manufacturers are using slightly different chips inside hubs marked with the same product name...

aciani, 2017-08-08:
Along the same lines, I concluded that I would likely need to modify my own hub, so I just bought a Link Depot 4-port hub (LD-USH-4POW) with mechanical switches.  I figured I can connect the source and drain (or collector and emitter) across the switch for an easy patch.  I decided to first test the hub, as it advertises per-port over-current detection and protection.

It made an audible squelch when it was plugged in, a bad sign: the filter to remove lower harmonics from the crystal oscillator is missing.  I popped open the case and saw that the board had positions for capacitors, resistors and diodes to filter in-rush power, sense current and filter the oscillator, but the components were missing.  Many of the solder joints appeared cold, and the solder seemed to be a lead-tin mixture (not RoHS compliant).  And surprise, the controller chip only supports ganged over-current detection and protection (I had to modify uhubctl to recognize ganged power switching hubs).

This Link Depot hub was clearly designed to be a proper hub, and may have been so at one time, but parts have been eliminated from its manufacture.  Does Link Depot even know?  Safe to say, I am not modifying this hub.

The missing parts and use of lead-free solder would only add about $0.30 to the manufacturing cost.  With overhead and profit margins, that means the $9.99 hub becomes a $10.99 hub.  Or maybe, the $9.99 hub becomes an $8.99 hub, and everybody pockets some extra dough.  Or maybe at the Chinese sweatshop end, the $2.00 hub becomes a $1.70 hub, and they pocket an extra $0.30 while cheating everyone up the line, because no one checks.

So it is no surprise at all that hub models which used to work might no longer work.  Consider that all of these hubs violate the USB 2.0 specification, the terms for licensing the USB name and logo, falsely advertise their characteristics (both electronically and on their packaging) and expose our motherboards to real risks of over-current damage.  Somebody hasn't been doing their job at the USB Implementers Forum.

My concern now is that the $39.99 hub is built identically to the $9.99 hub.  The Amazon and Anker hubs are those more expensive hubs.  A lot of them look externally identical, and even have similar packaging and text.  I am going to try some more of those more expensive hubs and find out.

If I find one that is mod-worthy, I will report that as well.

mvp, 2017-08-09:
Excellent thoughts, @aciani! This is good business idea: invest extra $1 into hub that supports power switching properly, brand it as premium USB hub that can control power on all ports, charge extra $10 from customers. I would totally buy it!

aciani, 2017-08-10:
It does seem like a good business opportunity, except for a significant marketing hurdle: many of those non-compliant, incompatible hubs already make the same or similar claims on their packaging.

For example, "Will not charge an iPad, but will link to one."  Could imply that the hub is capable of switching power.  Nope, the iPad will happily charge and over-draw current.

Then there is the slightly more common and overt, "Per-port over current detection and protection."  That sounds like the hubs can report power usage and cut power to individual ports.  No, everything is hard-wired to the bus, and there is not even detection much less protection.

And then there is the blatant claim of, "Per-port power switching."  If only they did, this wouldn't be so difficult.

mvp, 2017-08-19:
I don't see anything actionable on this issue from uhubctl standpoint. It works when hardware supports it. Closing.

closed 2017-08-19 by mvp