mvp/uhubctl #52: Not working on Raspberry Pi 3

smoneck, 2017-10-02:
```
$ sudo ./uhubctl -a off -p 5                 
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 5: 0103 power enable connect [0eef:0005 WaveShare WaveShare Touchscreen 2016-11-06]
Sent power off request
New status for hub 1-1, vendor 0424:9514, 5 ports
   Port 5: 0000 off
```
But the power stays on. :/

SlySven, 2017-10-02:
Duplicate topic, see #35!

You need to use port __2__ as the RPis use a ganged arrangement and all the USB ports are tied to THAT control line only. Incidently, do __not__ use port __1__ if you are SSH into the RPi via ___Wired___ Ethernet - that has the not well documented effect of shutting down that part of the combined Chip on at least the RPi B+, 2B and 3B models which implement this ganged power switching.

mvp, 2017-10-02:
Perhaps we should mention this in README.

smoneck, 2017-10-02:
Indeed, the USB-ports switch off with `-p 2`. Can you tell me how to get the port numbers? Or a `--list-ports` option? Congratulations! The first working way of shutting down the USB-ports and let Ethernet running! :-) `hub-ctrl` doesn't do anything working here ...

But I just tried `-p 1` and it seems to kill Ethernet and WiFi together ...

mvp, 2017-10-02:
@smoneck: yes, this is exactly how Raspberry Pi hardware works:
Port 1: also provides power to Ethernet and Wifi. Turning it off would kill network.
Port 2: operates in ganged mode, and would turn off/on power for all USB ports.
Other ports are not relevant for real operation, but would be displayed by `uhubctl`.

To enumerate all ports, simply run `uhubctl` without any options.

mvp, 2017-10-02:
Added a note about this in README.md in e53d6d2.

closed 2017-10-02 by mvp

smoneck, 2017-10-03:
Thank you very much!

Just final question: You write that USB port 1 is wired to control Ethernet/Wifi for just some models. I naively thought that all RPi3b are identical. Am I wrong?

mvp, 2017-10-03:
It says on some models because uhubctl works on B+, 2B, 3B models, but not all of these have port 1 wired to Ethernet (3B does).

mvp, 2017-10-03:
Ok, thanks for footnote idea. I'll try to do that :)

SlySven, 2017-10-04:
What __I__ meant when I wrote what I did is that the USB ports are __not__ controllable on RPi Model 1 B's, it was introduced on the 1+ which had a more efficient power supply circuit overall anyhow.

This is disappointing for me as the unit I do want it for is a 1B that will be running headless (it has a broken HDMI port) with a [UPiS](http://www.pimodules.com/_pdf/UPiS%20Module.pdf) RPi UPS Module which is not happy to supply more than 500mA when it needs to draw an additional 200mA to charge it's internal LiPo battery; sadly that is pretty close to the total when the 2x RFID Readers that are intended to control the lock on my home's front door are both powered up; I wanted to be able to power them down when not needed or when `init` gets a `POWERFAIL` message down its `/run/initctl` pipe or after suspicious (i.e. an _unknown_ tag is detected) but I cannot do that on that old an RPi model.

I was not certain about the wiring of the port 1 line though - I was extrapolating from my experiments with that 1B and another one (a 2B) but I do not have the other models to check it on.