mvp/uhubctl #54: Support for USB Pluggable 3.0 7-PORT CHARGING HUB

hursevich, 2017-11-08:
Is it supprted http://plugable.com/products/usb3-hub7bc/ ?

mvp, 2017-11-08:
I don't know - I don't have access to all the hubs available on market. I would appreciate if you try and report - I will add it to compatible devices table if it does work.

mvp, 2017-11-14:
Looking at internal design and chipset used for Plugable hub US3-HUB7C http://amazon.com/dp/B00RM5OF70, it looks internally identical to AmazonBasics 7 port hub that is listed as supported - they both use two 4-port daisy chained hubs based on VIA811 chipset.

The only difference seems that Plugable offers official firmware updates, which should make it better choice than AmazonBasics if it actually supports per-port power switching.

There are even mixed reports that some people were able to update their AmazonBasics hubs firmware to images supplied by Plugable.

I think it is worth checking Plugable hub. But, I will close this issue if I don't get any reports about it in about a week.

mvp, 2017-11-17:
Closing as not actionable. Feel free to reopen if you can confirm this hub working.

closed 2017-11-17 by mvp

mvp, 2018-02-18:
@hursevich: Can you please try this one again using uhubctl built from master branch?
Be sure to turn off power for both USB2 and USB3 virtual hubs. Thanks!

Julio-Guerra, 2018-03-01:
Here is what is see with the plugable USB3-HUB7BC and one device on port 1:
```
$ uhubctl
Current status for hub 20-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 20-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-10 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect []
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 20-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 highspeed power enable connect []
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

How do you turn off the usb 2 virtual hub? They have the same location `20-1`.

```
$ uhubctl -l 20-1 -p 1 -a off
Error: changing port state for multiple hubs at once is not supported.
Use -l to limit operation to one hub!
$ uhubctl -l 20-10 -p 1 -a off
Current status for hub 20-10 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect []
Sent power off request
New status for hub 20-10 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect []
```

Julio-Guerra, 2018-03-01:
@mvp I tested uhubctl with three distinct Plugable USB3 HUBs, on my macbook pro retina late 2013, with macOS 10.13.3, libusb@HEAD and uhubctl@HEAD.

They are based on two distinct VIA chipsets: VL 811 and VL 813. The result is quite unexpected to me: it turns off only after ~5 seconds, but then turns back on alone after a few seconds off. The state read by uhubctl is still off and the cascaded chipset disappears.  I need to turn it on again to see it again and be able to do new off/on requests.

Here are the logs for the VL 813 hubs but they all behave the same (USB3-HUB7BC (two cascaded VL 811), USB3-HUB7-81X and USB3-HUB7C2C (two cascaded VL 813)):

```
$ uhubctl
Current status for hub 20-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-10.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 highspeed power enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect

$ uhubctl -l 20-10 -p 1 -a off; uhubctl -l 20-1 -p 1 -a off
Current status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Sent power off request
New status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 highspeed power enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Sent power off request
New status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off

$ # ~5 secs before it actually turns off, but comes back on after a few seconds off
$ # what I see then:
$ uhubctl
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 5gbps SS.Disabled
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect 

$ # if I turn it on, it immediately turns power off, and comes back on after ~5 seconds
$ uhubctl -l 20-10 -p 1 -a on; uhubctl -l 20-1 -p 1 -a on
Current status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 5gbps SS.Disabled
Sent power on request
New status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02e0 5gbps power Polling
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
Sent power on request
New status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0101 power connect [2109:2813]

$ # the second chipset is listed again:
$ uhubctl
Current status for hub 20-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-10.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 highspeed power enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-10 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
```

mvp, 2018-03-01:
Check issue #33. Basically, on Mac, you need to build libusb from head.

mvp, 2018-03-01:
Sorry, I read it too quick, you are already on libusb head. Yes, this hardware is behaving weird. Perhaps turn off order matters? Or, maybe you need to power of both USB2 and USB3 very quickly?

mvp, 2018-03-01:
I reread your logs again, and I think you're doing it incorrectly.

This 7-port hub is designed as 2 4-port daisy chained together, which wastes 1 port on one of those hubs to connect them together. Note that you are trying to turn off this very port, which is special - it's only purpose is internal connection, you should never try to control power on it.

Julio-Guerra, 2018-03-01:
@mvp wow... I didn't fully understand the output of uhubctl until you said that... Ok, so in fact it works perfectly 😄Port 1 of the first chipset was indeed the nesting port...

I fully tested the USB3-HUB7BC (VL811-based), USB3-HUB7-81X and USB3-HUB7C2C (VL813-based) and they work with uhubctl.

It seems that when attached children are not fully suspended, the power comes back on, probably as a HUB feature when anything happens in the USB line (like a wakeup). I tested with advanced USB devices that use data and provide several device endpoints, and it does the same by partially going off. So basically, the devices should be recursively suspended to be able to power off the parent's HUB port.

Thank you !

mvp, 2018-03-01:
I have added this hub to supported devices list.

This whole USB3 story sounds awesome. I believe a lot, if not most, USB3 hubs actually should work fine with uhubctl.

Before we were struggling mostly because we didn't have proper USB3 support.
Now, many hubs we rejected before we find actually work just fine.

mvp, 2018-03-01:
@Julio-Guerra, I added USB3-HUB7-81X to the list as well.
However, I could not find any references for USB3-HUB7C2C anywhere.
Is it the same item as USB3-HUB7-81X?

Julio-Guerra, 2018-03-06:
US3-HUB7C : https://plugable.com/products/usb3-hub7c/ The model name I gave is the one at the bottom of the hub (I think it means "2 Charging ports").

USB3-HUB7-81X : https://plugable.com/products/usb3-hub7-81x/

USB3-HUB7BC : https://plugable.com/products/usb3-hub7bc/

mcba, 2018-06-03:
I don't understand how any VL813-based hub can have individual port control using uhubctl. 

From the datasheet for the VL813, the only power control pins are for over-current protection in gang mode (i.e., controlling all four ports at once) and individual control of one USB charging port. A hub schematic I found on the web had both these pins unconnected.

I bought a Volans VL-HC41 hub based on the VL813, and although it _appears_ to work with uhubctl (i.e., one can turn on/off the power reported by uhubctl for each port), it does nothing to the actual VBUS pins on the ports.

I opened up the hub and had a look, and all four VBUS pins are connected together, so the VL813 can't control the individual port power. In fact, the VBUS pins were connected to the external supply and the 5V input to the VL813, so gang mode power-down isn't possible either.

So should we be skeptical of reports that VL813-based hubs work with uhubctl?

mvp, 2018-06-03:
I guess we should be always skeptical about any hubs. It's quite common that hub reports supporting PPPS in hub descriptor, when they actually lack physical cirtuitry to do so. If you are sure these hubs don't do it right, we should remove them from supported list...

mcba, 2018-06-03:
Interestingly, the VL813 chip has 8 pins that are nominally for a green and amber status LED for each of the four USB ports. According to the datasheet these can be controlled "manually", which I guess means via the appropriate software. On my VL-HX41 hub, the green LED pins are unconnected, and the amber ones appear to be used for an SPI interface to flash memory that hasn't been installed. In any case, it would be relatively easy to use these pins as general purpose logic outputs to enable high-side switches to control the power to each USB port. A small daughter board and some surgery on the VBUS inputs to the ports (between the polyfuses and the VBUS pins) would do the trick. In fact, you could possibly short the fuse to ground as a crude way of cutting the power, and rely on the fuse to recystallize to turn the power back on - this is probably not a good idea...

I haven't been able to find any information on programming the pins (labelled USBHGRN1-4, and USBHMBR1-4). If the pins could be controlled, I would seriously look into building the daughter board.

mcba, 2018-06-03:
It appears that more recent VIA chips, such as the VL817, _do_ explicitly provide power enable pins for each port. It is then up to the OEM whether they use these pins. I have sent an email request to Orico (who make at least one hub using the VL817) to ask whether they support per-port power switching.

mvp, 2018-06-03:
As far as I know, USB 3.2 spec actually requires proper PPPS support - may that's why all USB vendors are starting to do it right...

mcba, 2018-06-03:
Instructions on how to turn on the per port LEDs are in the USB 2.0 specification document. I hacked into uhubctl.c to try this, and did manage to turn on one green LED pin on port 1 on my Volans VL-HX41 with VL813. So, this looks like a viable way of controlling the power per port, with some additional electronics added.
I guess that even though the VL817 and later chips support PPPS, manufacturers aren't going to want to spend the extra few dollars adding the power switching components, particularly as almost no software makes use of it.