mvp/uhubctl #55: [bug] Process hangs in D state on AmazonBasics hub after some time

szymonmatuszczak, 2017-11-10:
I've got AmazonBasics hub from the support list. Everything was perfect, but sometimes the uhubctl process hangs in D state and nothing can be done, even unplugging and plugging in again the hub does not help, the only help is the following procedure: unplug hub, reboot, plug hub (and it works for some time and hangs again). In the next week I'll replace my hub to old, but good 7-port D-Link and I'll let you know what is going on with other hub. I've got three devices connected to hub, drawing ~30mA power each, the same problem with and without external power. Same when compiling binary for i386 and amd64 (always amd64 kernel). My hardware and software info:

root@somehost:~# cat /proc/cpuinfo
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 15
model		: 111
model name	: AMD Sempron(tm) Processor 200U
stepping	: 2
cpu MHz		: 999.827
cache size	: 256 KB
fpu		: yes
fpu_exception	: yes
cpuid level	: 1
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 syscall nx mmxext fxsr_opt rdtscp lm 3dnowext 3dnow up rep_good nopl extd_apicid pni cx16 lahf_lm extapic cr8_legacy 3dnowprefetch vmmcall
bogomips	: 1999.65
TLB size	: 1024 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management: ts fid vid ttp tm stc 100mhzsteps

root@somehost:~# lsusb
Bus 001 Device 022: ID xxxx:xxxx Some_device_connected_to_hub
Bus 001 Device 016: ID 2109:2811 (first and internally daisy chained smart hub)
Bus 001 Device 020: ID xxxx:xxxx Some_device_connected_to_hub
Bus 001 Device 021: ID xxxx:xxxx Some_device_connected_to_hub
Bus 001 Device 014: ID 2109:2811 (first and internally daisy chained smart hub)
Bus 001 Device 002: ID 0781:5572 SanDisk Corp. (pendrive with OS)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

root@somehost:~# lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD/ATI] RS690 Host Bridge
00:01.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RS690 PCI to PCI Bridge (Internal gfx)
00:06.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RS690 PCI to PCI Bridge (PCI Express Port 2)
00:07.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RS690 PCI to PCI Bridge (PCI Express Port 3)
00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB600 USB (OHCI0)
00:13.1 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB600 USB (OHCI1)
00:13.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB600 USB (OHCI2)
00:13.3 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB600 USB (OHCI3)
00:13.4 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB600 USB (OHCI4)
00:13.5 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB600 USB Controller (EHCI)
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 SMBus Controller (rev 14)
00:14.1 IDE interface: Advanced Micro Devices, Inc. [AMD/ATI] SB600 IDE
00:14.2 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 Azalia (Intel HDA)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD/ATI] SB600 PCI to LPC Bridge
00:14.4 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 PCI to PCI Bridge
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] HyperTransport Technology Configuration
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] Address Map
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] DRAM Controller
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] Miscellaneous Control
01:05.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RS690M [Radeon Xpress 1200/1250/1270]
08:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 02)

root@somehost:~# uname -a
Linux somehost 3.2.0-4-amd64 #1 SMP Debian 3.2.78-1 x86_64 GNU/Linux

root@somehost:~# dpkg -l | grep libusb
ii  libusb-1.0-0:amd64              2:1.0.21-1                     amd64        userspace USB programming library
ii  libusb-1.0-0:i386               2:1.0.21-1                     i386         userspace USB programming library
ii  libusb-1.0-0-dev:amd64          2:1.0.21-1                     amd64        userspace USB programming library development files
ii  libusb-1.0-0-dev:i386           2:1.0.21-1                     i386         userspace USB programming library development files
ii  libusb-1.0-doc                  2:1.0.21-1                     all          documentation for userspace USB programming

root@somehost:~# dpkg -l | grep gcc
ii  gcc                             4:6.3.0-4                      amd64        GNU C compiler
ii  gcc-4.6                         4.6.3-14                       amd64        GNU C compiler
ii  gcc-4.6-base:amd64              4.6.3-14                       amd64        GCC, the GNU Compiler Collection (base package)
ii  gcc-6                           6.3.0-18                       amd64        GNU C compiler
ii  gcc-6-base:amd64                6.3.0-18                       amd64        GCC, the GNU Compiler Collection (base package)
ii  gcc-6-base:i386                 6.3.0-18                       i386         GCC, the GNU Compiler Collection (base package)
ii  gcc-6-multilib                  6.3.0-18                       amd64        GNU C compiler (multilib support)
ii  gcc-multilib                    4:6.3.0-4                      amd64        GNU C compiler (multilib files)
ii  lib32gcc-6-dev                  6.3.0-18                       amd64        GCC support library (32 bit development files)
ii  lib32gcc1                       1:6.3.0-18                     amd64        GCC support library (32 bit Version)
ii  libgcc-6-dev:amd64              6.3.0-18                       amd64        GCC support library (development files)
ii  libgcc1:amd64                   1:6.3.0-18                     amd64        GCC support library
ii  libgcc1:i386                    1:6.3.0-18                     i386         GCC support library
ii  libx32gcc-6-dev                 6.3.0-18                       amd64        GCC support library (x32 development files)
ii  libx32gcc1                      1:6.3.0-18                     amd64        GCC support library (x32)

mvp, 2017-11-10:
All that uhubctl does is to send low level hardware signal to the hub turn off specified port.

Process in D state means that process called into kernel, but it did not return, and thus process cannot be killed (see more in https://unix.stackexchange.com/q/364100/199308).

Why it ended up in D state? I don't know, but I think most likely reason it happened when uhubctl called `libusb_control_transfer()` here:

https://github.com/mvp/uhubctl/blob/e53d6d2f367504dc9c10d8e2c084aa04e97cbd6a/uhubctl.c#L635-L639

It might be related to this discussion in linux-usb mailing list: https://www.spinics.net/lists/linux-usb/msg157413.html

Basically, kernel USB driver is aware that power feature is being disabled on given port, but it does not necessarily remove attached devices from kernel USB stack.

Suggestions were made to have kernel's `hub.c` to handle this properly https://www.spinics.net/lists/linux-usb/msg157435.html, but no active decision was made on this one.

Another possible issue is that your hub hardware does not properly respond to port turn off, and stops communicating to host at all. I would say that this is a bug in kernel as well - no matter how badly external hardware is behaving, host should not be confused, and kernel USB stack should be able to recover, in worst case by forcibly disconnecting misbehaving hardware.

Note that most (if not all) USB3 hubs have upgradable firmware, and your AmazonBasics hub is no exception, which might just fix your problem. Can you try to upgrade hub firmware? I believe you will need Windows to do it.

szymonmatuszczak, 2017-11-12:
I will make some tests on this week on identical "staging setup" after replacing the hub on the production setup. I am kind of perfectionist so you can be sure that I will check every case in relation to this hub :) Please do not close the issue yet.

mvp, 2017-11-12:
If you have access to other supported OS, e.g. Mac or FreeBSD, it would be interesting to check if changing host from Linux to Mac would make your hub work better.

szymonmatuszczak, 2017-11-19:
Argh... after few hours of debugging i have multiple clues... the first is that compiling with gcc-4.6 doesn't really change anything. I've made few test cases, like "while true; do uhubctl; done" and "while true; do uhubctl -l 1-8.3 -p 1 -a cycle -d 2; sleep 5; done" ran in screen from root... and it seems to be OK without any changes - even after few hours run-time any process does not remain in D state. I've also noticed, that my scripts was executing uhubctl binary very often, and sometimes simultaneously from few threads at one time - eliminated this, updated both VIA hub chipsets firmware to newest from Pluggable website for VL811+ chipset (strange situation: out-of-the-box one of chips was programmed with firmware 9084 and the second with firmware 9081, now both are programmed with firmware v 9095) and it seems to be OK, but as you've stated this is really a kernelspace bug, and even if we are able to hang the USB hub by sending too much requests, the kernel should be able to handle this. Simultaneously, I am testing the $8 (used) silver D-Link DUB-H7... i will post all misbehaviors I find. Maybe I'll also test one of hubs on other Unix-like OSes in the nearest future :)

szymonmatuszczak, 2017-11-19:
The other issue with the luckless AmazonBasics hub... after few thousands powercycles of some ports and/or >~20h hub uptime, some devies are "vanishing" from uhubctl (sometimes inside-daisy-chained hub, of course I am NOT touching its power) until reboot. I didn't found out this bug on the DUB-H7 (yet).

mvp, 2017-11-19:
I'm glad that upgrading hub firmware fixed it!

I have used D-Link DUB-H7 quite extensively in the past. In my experience, 95% of hubs will work with uhubctl flawlessly for months and years. But some could be behaving very weird, and nothing but replacing could solve it.

Closing as not actionable from uhubctl perspective.

closed 2017-11-19 by mvp