mvp/uhubctl #56: A list of incompatible devices would be useful

mleguen, 2017-11-17:
As most USB devices are incompatible, maintaining a list of incompatible devices, and suggesting people to report them would also be useful. I could help with initializing it.

This would help people limit the number of devices to buy and test before finding one which works (something I am doing at the moment).

A `--report` option dumping ids of all found hubs (smart and classic) could also be something useful to develop and ask people to add output to issues when reporting. I could also help with that.

mvp, 2017-11-17:
Maintaining list of incompatible hubs is futile - that would be 99% of hubs on market.

It's easier to maintain only list of certainly compatible hubs instead.

As for dumping ids, if you run `uhubctl` without any options, it will by default display all compatible hubs.

mvp, 2017-11-17:
Please feel free to run uhubctl on any hubs you have access to, and report if they are detected by uhubctl.

However, not much else can be done from development point of view. Closing as not actionable.

closed 2017-11-17 by mvp

SlySven, 2017-11-18:
🤔 Um, from a basic consideration of what I think is called _set_ theory the issue that people have with the status quo is determining whether the hub that _they_ have or are considering is in the set of _untested_ for compatibility with `uhubctl` as opposed to being in the _broken_ set.  Whilst both of these are much larger than the set of those hubs in the _working_ one - **they are not the same** - so I think it is needed to maintain a list of members of **two** of the three sets not just _one_...

Given that the _untested_ set is inherently ill-defined the only way to record for users whether what they have got is likely to work **or not** is to record the _broken_ hubs results as well as the _working_ ones.  Perhaps this could be recording in a table in the Wiki area here on Github.  Given that there are also an undefined number of manufacturers I guess it ought to be able to be sorted by USB Vendor/Product ID and given that those apply to the chip and not necessarily the circuit into which it is put - even that classification may have to be sub-divided into the brand name/model number/revision number in order to pick out the fine details of those hubs that have got it right!

Don't forget, if we do not record the _broken_ results it will only lead to different potential users having to test the unit they are considering themselves which is a waste of group effort - whilst negative results are not as good news as positive ones they are still data items that ought to be retained simply because they infinitesimally chip away at the _unknown_ set whereas ignoring them leaves it unchanged - and does not record/reward at all the effect the tester made in trying a sample of the hub concerned. 🙂

mvp, 2017-11-18:
@SlySven, if people want to check if hub that they have works, all they need to do is run uhubctl and see if it is detected - what could be easier?

Maintaining list of all hubs ever manufactured is virtually impossible task. It's similar to task of maintaining list of all USB vendors, and requires having large database. It will be extremely difficult to ensure that it is correct.

SlySven, 2017-11-18:
> Maintaining list of all hubs ever manufactured is virtually impossible task...

Agreed - but that is NOT what I was saying - what I was saying was something about "Maintaining a list of all hubs that have been **tried**" and the results both positive and **negative**!

Though I am rapidly coming to realise that the USB ID values are not necessarily going to be enough to say "that combination of Vendor::Product is/is not going to work" as the same chip may be used in different implementations and they can differ at the individual manufacturer's (or Far Eastern cloner's) whim (or willingness to spend the extra cents on the additional hardware) to create one that actually comes up to USB specifications... 🙁