mvp/uhubctl #57: AmazonBasics 7 Port USB 3.0 Hub with 12V/3A Power Adapter no longer compatible

mleguen, 2017-12-04:
I bought an AmazonBasics 7 Port USB 3.0 Hub a few weeks ago, as listed in top of the compatibility list, and it is not compatible.

It is listed as 2109:2811, as in the compatibility list (ASIN B00E4J7PTM).

It does say it can switch power per port but does not do it.

mleguen, 2017-12-05:
> Note that hubs are sorted not by best working, but simply by name.
>
> That said, other users have reported this hub as working before.
> Can you please provide more information as to what exactly doesn't work for
> you?
>
>    1. Which host OS you are using? Linux, Mac, FreeBSD? Note that on Mac OS
>    Sierra+, you have to compile libusb from HEAD for it to work.
>    2. Does uhubctl (without any parameters) show status for this hub? It
>    should show 2 4-port hubs, daisy chained to each other. You need to be very
>    careful when specifying hub and port to control power on.
>    3. What happens if you try to turn off/on USB port on this hub? What
>    device is connected to the port you are trying to turn off/on? Did you try
>    different devices to control power on?
>    4. Some hi-speed devices need to use -r N option to actually turn power
>    off (N is some number).
>
> <https://github.com/mvp/uhubctl/issues/57#issuecomment-349148310>, or mute
> the thread

Answers to your questions:

1. I am using Ubuntu (the latest LTS).

2. Yes, the hub is shown the way you describe it.

3.1. Using a real USB device (wifi dongle with a LED indicator): when I
turn off the port the device still works. After unplugging and replugging
the device it no longer works. When I turn on the device starts working.

 3.2 Using a dumb device only drawing power from the USB port (charging
only USB cable with no data pins connected to my phone): nothing happens at
all, it is always charging, even after turning the port off and unplugging
and replugging the device.

In my opinion, the real device does not work after tuning the port off,
unplugging and replugging, only because the hub considers it to be in power
off state, so it does not start talking with it.

4. No high speed device used.

mleguen, 2017-12-05:
I asked a question on Amazon before buying this hub and received an answer
a while after (hub already received): someone else also considered this hub
as not per port power switching  compatible.

mvp, 2017-12-05:
It's really confusing when some users report hub as working well, and some like you say it doesn't work.

Don't get me wrong - I believe that version you purchased doesn't work. At the same time, exact same model has worked for someone else who reported it few months ago.

I am torn apart whether to remove it from compatibility list... Perhaps I should say that some versions of AmazonBasics hub are known not to work?

I have personally used few hubs from compatibility list that definitely do work. I can guarantee that D-Link DUB-H7 silver edition does work - I own few of them. EVB-2517 also works - I have it in my possession. As for most other hubs in compatibility list, they were reported by various people who compiled uhubctl and tried them on different hardware.

In general, I only add new hardware to compatibility list when at least 2 independent sources report it as working.

However, there are plenty of examples when seemingly exact same product actually has different chip or chip configured differently. In this case, it's possible that AmazonBasics has saved few cents by removing physical power switch...

mleguen, 2017-12-06:
It may be because several factories are producing the same model, with
different understanding of the specifications. And such a feature is
probably never tested afterwards because most clients don't care about it.

A way to deal with such models could be to add a "Compatibility" column
with levels such as "Always", "Sometimes" (or even "Never", but we already
talked about this point).

LoganLechat, 2018-01-25:
Hi, 
Just order same USB HUB part number B00E4J7PTM chipset 811 and the power stay on.

Thanks

mvp, 2018-02-03:
Added a warning note to supported devices list in 022e321. Closing this issue.

closed 2018-02-03 by mvp