mvp/uhubctl #58: Supported: AmazonBasics USB 3.0 4 port working on USB 2.0 host only

al177, 2017-12-18:
This Amazon branded 4 port USB 3.0 hub is working with uhubctl provided that it is connected to a USB 2.0 host.  Defaults for cycle work for individual ports and all ports.

But on a USB 3.0 host, "-a off" seems to reset ports but not power cycle them.  I have tried various values of "-r N" and "-w N" and resetting the hub, host port, and replugging the hub with no effect when on a USB 3.0 host.

The chipset is VL811+, the HS device is 2109:2811, the SS device is 2109:8110, and the factory firmware is 90.74.  Upgrading firmware to 90.95 through Plugable's VL811+ updater (https://plugable.com/drivers/hubfirmware/) makes no difference in behavior.

mvp, 2017-12-18:
Interesting. It looks like this hub contains different chips for USB 2 and USB 3 operation.
And it looks like those chips are almost completely independent, and have different level of support for PPPS for USB2 vs USB3 mode.
This might explain issue #57 (or maybe it is even a duplicate).

Regarding cycle vs power off - it should not be a whole lot of difference. Cycle simply means power off, wait some delay (2s by default), then power back on. Off means only power off.

Unfortunately, not much can be done by `uhubctl` to fix broken hardware. But I will add a note for AmazonBasics about this caveat - thank you so much!

al177, 2017-12-18:
I've opened the hub, and it has only one hub controller.  The ICs I can find:

VL811-Q8P - Via hub
MXIC25L512E - 512Mbit SPI flash (firmware for VL811)
2x XR77103 - Exar switching regulators
4x GMB51001E2P11U  - GMT USB power switch (marked G547E2)

Plus a few unidentifiable SOT-3 devices.

LoganLechat, 2018-01-25:
Hi
Just tested the same USB HUB and USB2 or USB3 host, the power are still active on the ports. 
Thank you

al177, 2018-01-25:
@LoganLechat In your USB2 testing, is that a USB2 hub port, or USB2 root port?

LoganLechat, 2018-01-25:
@al177 Directly connected on my Gigabyte 2807 with USB2 and USB3 ports. 
tested on both

harjotcheema, 2018-01-26:
I also tested Amazon Basics HU3641V1, 4 Port USB 3.0 Hub on both USB2.0 and USB3.0 and it turned off power for the port as below:
uhubctl -a off -n 2109 -p 1234
Current status for hub 20-2, vendor 2109:2811, 4 ports
   Port 1: 0000 off 
   Port 2: 0100 power 
   Port 3: 0100 power 
   Port 4: 0100 power 
Sent power off request
New status for hub 20-2, vendor 2109:2811, 4 ports
   Port 1: 0000 off 
   Port 2: 0000 off 
   Port 3: 0000 off 
   Port 4: 0000 off 

The problem is the phone connected to the usbhub continues to charge even though the power has been turned off. Is there a way we can stop the phone from charging with usb power off?

mvp, 2018-01-26:
This must be hardware issue with AmazonBasics hub.

Please read this comment I wrote long time ago: https://github.com/mvp/uhubctl/issues/12#issuecomment-282239737

Basically, some hubs may omit proper power switches, despite USB chipset supporting power switching.

harjotcheema, 2018-01-26:
Can you recommend a hub that works.

mvp, 2018-01-26:
D-Link DUB-H7 old silver edition definitely works. 

Raspberry Pi B+, 2B, 3B internal USB hub also works, but it controls power for all USB ports.

mvp, 2018-02-03:
Added a warning note to supported devices list in 022e321. Closing this issue.

closed 2018-02-03 by mvp