mvp/uhubctl #59: Can I identify compatible switches by looking at the pcb?

jensolsson, 2017-12-30:
I guess there are some switch or mosfet etc. How does it usially look on compatible pcbs?

SquareBoot, 2017-12-31:
Hello,
maybe not fully related to this issue but I made some tests. On some hubs, like mine, power switching is just a software trick, so nothing about MOSFETs. See issue #60 if you're interested.
Kind regards,
SB.

mvp, 2017-12-31:
@jensolsson, I am not electrical engineer and frankly don't know much. I can only suggest looking at hubs that are known to work for sure. Most trouble free operation I've seen on D-Link DUB-H7 old silver model, and SMSC/Microchip EVB-USB2517 evaluation board. You can find full schematics for this board on Microchip website: http://www.microchip.com/DevelopmentTools/ProductDetails.aspx?PartNO=EVB-USB2517

jensolsson, 2018-01-01:
@SquareBoot in your case i think the hub is not compatible. I beliebe you need some kind of way to shut off the power for ppps to be correctly implemented. 
@mvp i am really looking for a usb3 hub. Matbe it is not available...

mvp, 2018-01-02:
@jensolsson, I don't know of any better information sources for USB2/USB3 hubs other than EVB-USB2517 I provided above.
If you find anything, please share.
Meanwhile, I don't think there is anything actionable from `uhubctl` standpoint on this issue.
Closing, but feel free to reopen.

closed 2018-01-02 by mvp