mvp/uhubctl #60: Trust OILA 7 port USB 2.0 hub

SquareBoot, 2017-12-31:
Hello,
I'm a newbie with `uhubctl` but I've made some tests with this powered hub, which seems to support correctly per-port power switching. However, only `hub-ctrl.c` recognizes it: using it I was able to turn on/off each port but no way to find it in `uhubctl`. I was wondering where the problem is since `uhubctl` was originally inspired by `hub-ctrl.c`.
Moreover, I found that this hub behaves in a strange way: it **does not turn off** the power to his slave devices (speakers or USB chargers will continue working) but, instead, it makes them "sleep" (I can still find them in `lsusb`, not in other applications). From the `hub-ctrl.c` readme: 

> This only works on USB hubs that have the **hardware** necessary to allow software controlled power switching.

So, how is power switching carried out? Slave devices, the hub's hardware/software or operating system? Since power is never turned off, why not all the hubs support it (like my laptop's root hubs)? Or, maybe, due to my hub hardware incompatibility, `hub-ctrl.c` is just disconnecting them from the operating system.
I verified this behaviour with several USB sticks, a camera and a phone and I made a map of all the ports, if it can help.
Kind regards,
SB.

mvp, 2017-12-31:
Most likely reason why uhubctl doesn't detect your hub is because it requires per-port power switching support. Detection is done in function is_smart_hub(). uhubctl requires that hub supports power switching per-port, and overcurrent protection per-port or ganged (all ports at once).

It is possible that your hub reports power switching ganged (all ports at once). Uhubctl currently doesn't detect or support such hubs. Even if uhubctl detects hub, it doesn't guarantee that it will actually work, because some manufacturers do not include physical power switches for it to really cut off power.

Current compatibility list was compiled by carefully collecting information about hubs that support true power switching. That said, some manufacturers may release newer products with exact same product code which may have different chips inside and power switching may not work for them... For example, D-Link DUB-H7 old silver model works perfectly, while newer edition has exact same model id and doesn't work...

SquareBoot, 2017-12-31:
> It is possible that your hub reports power switching ganged (all ports at once)

`hub-ctrl.c` reports "ganged" only on my laptop's root hubs, not on the external one.

> uhubctl requires that hub supports power switching per-port

It is per-port, but a sort of software trick, I mean. It only disconnects the device from the OS.

> that support true power switching

OK, so my hub is not supported. I'll use `hub-ctrl.c` keeping in mind that I'll only be able to connect and disconnect them, no power switching.
Thanks,
SB.

closed 2017-12-31 by SquareBoot