mvp/uhubctl #62: YepKit and Acroname Hubs

dbking77, 2018-01-08:
Have not tested, but website explicitly claims support for port power switching
- https://www.yepkit.com/products/ykush
- https://www.yepkit.com/product/300110/YKUSH3

Also not tested.  Not clear it is controlled by or standard PPPS interface, or through separate management interface. 
- https://acroname.com/store/s79-usbhub-3p?sku=S79-USBHUB-3P
- https://acroname.com/store/s77-usbhub-2x4?sku=S77-USBHUB-2X4

mvp, 2018-01-08:
As far as I can tell, these products use custom management channel to control port power, most likely based on USB HID protocol.

On other hand, `uhubctl` will control any hub that supports standard per port power switching as defined by USB spec, which doesn't require any extra channel or interface to control port power.

mvp, 2018-01-16:
I think it might be good idea to incorporate support for Acroname and Yepkit hubs to use exactly the same command line interface. But unfortunately it needs full understanding of their custom protocols, and probably linking to `hidapi` library.

I'm not excluding possibility of this happening if someone proposes a patch, but for now its out of uhubctl scope.

closed 2018-01-16 by mvp