mvp/uhubctl #64: RPI 3B shutting port 2 shuts down all ports

wreuven, 2018-01-16:
> sudo uhubctl
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect [0424:ec00]
   Port 2: 0100 power 
   Port 3: 0303 lowspeed power enable connect [045e:0083 Microsoft Basic Optical Mouse]
   Port 4: 0303 lowspeed power enable connect [0603:0002 SINO WEALTH USB Composite Device]
   Port 5: 0100 power 

> sudo uhubctl -a off -p 2
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 2: 0100 power 
Sent power off request
New status for hub 1-1, vendor 0424:9514, 5 ports
   Port 2: 0000 off 

> sudo uhubctl
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect [0424:ec00]
   Port 2: 0000 off 
   Port 3: 0100 power 
   Port 4: 0100 power 
   Port 5: 0100 power 

> sudo uhubctl
Current status for hub 1-1, vendor 0424:9514, 5 ports
   Port 1: 0503 highspeed power enable connect [0424:ec00]
   Port 2: 0000 off 
   Port 3: 0100 power 
   Port 4: 0100 power 
   Port 5: 0100 power

mvp, 2018-01-16:
Yes, this is quirk of Raspberry PI internal hardware design. USB port 1 is wired to control power on network (wifi+Ethernet), and USB port 2 is wired to control power on all remaining USB ports.

closed 2018-01-16 by mvp

Brainsway-Cloud, 2018-01-16:
Does this mean that I can control ports 3,4,5 independently?

mvp, 2018-01-16:
No, you cannot control ports 3,4,5 independently. Port 2 will alter state for all ports 2,3,4,5 at once.