mvp/uhubctl #65: Cypress CY7C65632 HX2VL USB2.0 hub IC working

al177, 2018-01-25:
I have been successfully using uhubctl for PPPC with a custom PCB containing the Cypress CY7C65632 4 port USB-HS hub IC.  This implementation is strapped for PPPC and does not use an external config ROM to set custom VID:PID or other parameters.  I would expect that other members of the Cypress HX2VL family will work the same.

The device ID is 04b4:6570 which is the Cypress assigned default.

SlySven, 2018-01-26:
Is that a custom PWB that you have designed/built yourself? I would ask if you'd consider the CY7C65642 with the quad-Transaction Translator compared to the single TT of the IC you mention as that will work better with USB 1.0/1.1 when used with a USB 2.0 upstream connection...

al177, 2018-01-26:
The application I'm using it in connects only to HS devices, so multi TT would be a waste of money.

SlySven, 2018-01-26:
> [The application I'm using it in connects only to HS devices, so multi TT would be a waste of money.](https://github.com/al177/terrible-pi) 👍

... OMG [Internet of _Pants_](https://github.com/al177/wifi_pants) 🤣 

I tell you what though - I think a PWB with either the CY7C65632 or 42 in a RPi Zero outline board and two/three or four ports could be very attractive to battery/solar operated projects that want power control on USB peripherals!

As it happens I have a RPi B+ project where I would like such a thing myself - a Home Automation systems where I have a couple of low cost USB RFID readers and I want them for input to a state machine that will unlock my front door under the right circumstances.  Given that the whole project is run from a 12V battery charger continuously connected to a Lead-Acid battery (as a primary UPS) powering an RPi specific (secondary) UPS and that there are situations where I want to turn the RFID readers _off_ {like when the mechanical door deadlock on the [AV2-bluematic](https://www.winkhaus.com/en-gb/~/media/files/uk/documents/stv_autolock_av2_and_bluematic_av2e_br_032013_uk.pdf) is thrown which blocks the electric unlock - or when Mains Power fails/the pre-payment meter runs out 😊 } I would be interested if the price point is right - hell I might go for self assembly - though hand soldering of SMD passives is a pain by hand...

mvp, 2018-01-26:
Thanks for the update, @al177! I will add a note about it to README when I have a chance, perhaps this weekend.

mvp, 2018-02-01:
Added to supported list in a1b7411c49.

closed 2018-02-01 by mvp