mvp/uhubctl #66: USB hub model selection

get33, 2018-01-28:
I read all possible information about the hubs from your list of supported models, settled on two models (D-Link DUB-H7 and AmazonBasics 7 Port). Which of these two models can you advise?

P.S. found the information that on D-Link DUB-H7 the power does not turn off completely, but only the voltage decreases. is it so?

mvp, 2018-01-28:
Unfortunately, D-Link decided to keep exact same model name for many completely different revisions of their hardware.

D-Link DUB-H7 old silver edition (EOL 2010) works properly, but it can be hard to find it.

As far as I know, there are at least 2 more newer black editions of D-Link DUB-H7 - slightly older with 2 charging ports (doesn't work at all), and most recent with all identical ports which is recognized, but doesn't necessarily cut charge power.

Re AmazonBasics hubs, there are so many conflicting reports that it works for some people but not for others, I don't know if Amazon is releasing different revisions under the same name?

closed 2018-01-28 by mvp

SlySven, 2018-01-29:
Has anyone had any traction with explicitly asking for PPPS before they order a Hub and then forcing the retailer to take it back when the product turns out not to have it?

In the UK we had the [Sale of Goods Act](https://en.wikipedia.org/wiki/Sale_of_Goods_Act) (now supplemented with the [Consumer Rights Act](https://en.wikipedia.org/wiki/Consumer_Rights_Act_2015)) which _I believe_ required something being offered for Retail Sale to do what it said it did, and if you made it clear that there was a (reasonable?) feature that you needed in the product and then they sold you something which did not in fact have it you would be justified in asking for a refund.  OTO IANAL so YMMV :grin: