mvp/uhubctl #67: D-Link DUB-H4 do works

Manel3, 2018-02-01:


mvp, 2018-02-01:
@Manel3, which version of D-Link DUB-H4 are you talking about?
I have tested old silver edition of D-Link DUB-H4 (circa 2010), and it does not support PPPS.

Do you happen to have newer black version of this hub? Can you please provide any more identifying infromation, e.g. revision or model numbers printed on label. And, get you show output from uhubctl?

Also, please check if say phone connected to the hub stops charging when you turn power off using uhubctl?

Manel3, 2018-02-02:
Yes it is black new version.
P/N: EUBH4EB......D1E
H/W Ver.:D1
It is the one which appears when you search D-Link DUB-H4 in Amazon.

uhubctl
Current status for hub 3-6, vendor 05e3:0608, 4 ports
   Port 1: 0100 power 
   Port 2: 0100 power 
   Port 3: 0100 power 
   Port 4: 0100 power

uhubctl -l 3-6 -a 0
Current status for hub 3-6, vendor 05e3:0608, 4 ports
   Port 1: 0100 power 
   Port 2: 0100 power 
   Port 3: 0100 power 
   Port 4: 0100 power 
Sent power off request
New status for hub 3-6, vendor 05e3:0608, 4 ports
   Port 1: 0000 off 
   Port 2: 0000 off 
   Port 3: 0000 off 
   Port 4: 0000 off

I confirm I can control phone charging.

mvp, 2018-02-02:
Very interesting! Awesome news, I will add this one then.

What's really weird is that for old silver edition, DUB-H7 worked when DUB-H4 didn't. Now for new black edition DUB-H7 doesn't seem to work, when DUB-H4 does. Go figure :)

Manel3, 2018-02-02:
Note: In amazon.com there are two different models, mine is the one with all USB connector white (the available via prime) in fact I bought this:
https://www.amazon.es/gp/product/B00006B7DA/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

Manel3, 2018-02-02:
I guess if windows port of uhubctl worked it would be an important motivation for makers to build/keep this feature.

mvp, 2018-02-02:
libusb 1.0.22 should include alternative backend driver on Windows: usbdk.

I would expect that uhubctl might work on Windows if usbdk backend is selected upon libusb initialization.

mvp, 2018-02-02:
Added in 91e987527.

closed 2018-02-02 by mvp