mvp/uhubctl #68: AmazonBasics hubs work only when plugged to USB 2.0 upstream port

mvp, 2018-02-03:
There are many conflicting reports that AmazonBasics hubs work or don't work with uhubctl. It seems that it only works if  hub is plugged to USB 2.0 upstream port, and doesn't when plugged to USB 3.0.

Apparently, AmazonBasics hubs are designed as composite USB device for their USB2 and USB3 subcomponents. uhubctl detects only USB2 hub with id 2109:2811, not USB3 hub with id 2109:2810.

This could be an issue with libusb device enumeration, as uhubctl is using standard libusb primitives to enumerate all devices.

mvp, 2018-02-10:
I have obtained StarTech ST4300USB3 4-port hub with exact same vendor id as AmazonBasics: 2109:2811.
This hub has similar issue: I can turn off data connection with uhubctl, but connected phone continues to charge. However this condition is true regardless of connecting to USB2 or USB3 upstream port...

mvp, 2018-02-12:
I think I know what the problem is. Working on adding better support for superspeed USB 3.0 hubs...

mvp, 2018-02-13:
I have put preliminary support for USB 3.0 hubs in usb3 branch: https://github.com/mvp/uhubctl/tree/usb3, commit 8e667cde10fe9.
Unfortunately, I could not actually change power state for USB3 ports on StarTech ST4300USB3. Linux kernel seems to be completely ignoring USB control request to clear `USB_PORT_FEAT_POWER` on this USB 3 hub.
Can someone try this on other USB 3 hubs, perhaps AmazonBasics? Thanks!

mvp, 2018-02-13:
@al177, @LoganLechat, @mleguen, @hursevich, @aciani, @baclark774, @jensolsson, @phhusson: can you please try to build from usb3 branch https://github.com/mvp/uhubctl/tree/usb3 and see if you can control power for your USB3 hubs?
It looks like a lot of USB 3.0 hubs on market report supporting PPPS (maybe even all of them?), but I was not able to control power on StarTech ST4300USB3. This could be issue with Linux kernel driver though.
I need your input if this works on various models...

jensolsson, 2018-02-13:
I tried on an Orico hub I have on OSX, seem to accept the command but nothing happens. A bit awkward that it shows both USB-2 ports and USB-3 ports with the same location, so I need to specify both the vendorid and location for it to try to switch. Maybe there should be a -u 3 -u 2 to select between usb 2 or 3. I will try some more hardware tomorrow.

mvp, 2018-02-13:
@jensolsson: unfortunately, this is how USB3 works. It exposes the same physical hub as 2 independent devices: one is USB2, and another USB3. Actual device that was plugged into the hub port will appear to be connected to USB2 or USB3 hub depending on its connection speed.

And yes, we might want to add a switch to filter only USB2 or USB3 hubs.

mvp, 2018-02-13:
@jensolsson: if you see 2 hubs at the same location, you must be affected by issue #33. Simplest fix on OS X is to reinstall libusb from head:

    brew uninstall libusb
    brew install libusb --HEAD

and recompile uhubctl.

al177, 2018-02-14:
Branch usb3 behaves the same as master with the Amazon 4-port hub when connected to USB3.

mvp, 2018-02-14:
@al177, same means what exactly? You should see 2 virtual hubs: USB2 and USB3. For me, I can cut data connection on USB2 hub, but phone continues to charge. On USB3 hub, power off command is ignored by kernel or hub itself.

al177, 2018-02-14:
I see both hubs when connected via USB 3.0.  However when I force it to use the specific USB 2.0 device (via -l) nothing changes when I ask it to -cycle.

The same operation works when connected to a 2.0 host port.

mvp, 2018-02-14:
@al177, thanks!
At least on Linux, I'm afraid that USB 3 request to [turn port power off](https://github.com/torvalds/linux/blob/6556677a80400ca5744340d8ff6fbed22621293e/drivers/usb/host/xhci-hub.c#L1383-L1385) is completely ignored by xhci kernel driver in function [`xhci_set_port_power()`](https://github.com/torvalds/linux/blob/6556677a80400ca5744340d8ff6fbed22621293e/drivers/usb/host/xhci-hub.c#L570), specifically by checking if USB port is power manageable in ACPI [`usb_acpi_power_manageable()`](https://github.com/torvalds/linux/blob/6556677a80400ca5744340d8ff6fbed22621293e/drivers/usb/host/xhci-hub.c#L591-L595). I assume other operating systems like Mac might be doing the same check.

I think proving that may require compiling custom Linux kernel with this check disabled. Or, perhaps finding any host system where `usb_acpi_power_manageable()` returns true.

mvp, 2018-02-15:
Thanks to @Kyklas tip in https://github.com/mvp/uhubctl/pull/70,
there is a fix fb5d521290b, which makes it work for StarTech ST4300USB3 (apparently exactly the same internals as AmazonBasics hubs).
However, to remove power both USB2 and USB3 ports need to be powered off.
Perhaps we should create some kind of virtual merge between USB2 and USB3 counterparts?

# Examples

## Connected flash drive to port 4 of the USB 3.0 hub StarTech ST4300USB3:
```
$ ./uhubctl -n 2109
Current status for hub 4-3.2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 0203 5gbps power U0 enable connect [0951:1666 Kingston DataTraveler 3.0 50E549513780E0A1E94AE97E]
Current status for hub 3-3.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
## Powering off:
```
$ ./uhubctl -n 2109 -l 3-3.2 -p 4 -a 0
$ ./uhubctl -n 2109 -l 4-3.2 -p 4 -a 0
```
## Powered off:
```
$ ./uhubctl -n 2109
Current status for hub 4-3.2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 0080 5gbps SS.Disabled
Current status for hub 3-3.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
```

mvp, 2018-02-15:
Support for USB3 has been merged into master.
We need more testing against various USB 3.0 hubs.
Also need to add support for turning off/on ports for both USB2 and USB3 counterparts of the same USB3 hub.

mvp, 2018-02-17:
Can anyone confirm that AmazonBasics hubs do turn off power when both USB2 and USB3 ports are turned off (using USB3 upstream connection), when you build uhubctl from master branch?

In my test on StarTech hub, even if data connection is cut off, connected Android phone still continues to charge...

mleguen, 2018-02-17:
I got mine back yesterday, so I will try to test that this week-end.

<https://github.com/mvp/uhubctl/issues/68#issuecomment-366422879>, or mute
the thread

mleguen, 2018-02-18:
My AmazonBasics hub does not turn off power when both USB2 and USB3 ports are turned off.

Compiled from master branch.

I plugged a USB key to locate the port I was going to work with (here 3-1 and 4-1 port 1) :
```
$ sudo ./uhubctl 
Current status for hub 2-1 [8087:8000, USB 2.00, 8 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 1-1 [8087:8008, USB 2.00, 6 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 4-1.3 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 4-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 5gbps power U0 enable connect [8564:1000 JetFlash Mass Storage Device 03CXFHF2E93SRBJ8]
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 0263 5gbps power U3 enable connect [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 3-1.3 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 3-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 highspeed power suspend enable connect [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0100 power
```

I replaced it by a small USB lamp (not a smart device, only using power) on the same port.

I then switched the power off for port 1 for devices 3-1 and 4-1 :
```
$ sudo ./uhubctl -a0 -p1 -l 3-1
Current status for hub 3-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power off request
New status for hub 3-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
$ sudo ./uhubctl -a0 -p1 -l 4-1
Current status for hub 4-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
Sent power off request
New status for hub 4-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
$ sudo ./uhubctl
Current status for hub 2-1 [8087:8000, USB 2.00, 8 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 1-1 [8087:8008, USB 2.00, 6 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 4-1.3 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 5gbps power Rx.Detect
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 02a0 5gbps power Rx.Detect
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 4-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 5gbps SS.Disabled
  Port 2: 02a0 5gbps power Rx.Detect
  Port 3: 0263 5gbps power U3 enable connect [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 5gbps power Rx.Detect
Current status for hub 3-1.3 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 3-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0507 highspeed power suspend enable connect [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0100 power
```

The USB2 hub's port 1 is off and the USB3 one is disabled, but the USB lamp is still powered. It should have been switched off.

mvp, 2018-02-18:
This is unfortunate, and exactly matches behavior I'm seeing from StarTech USB3 hub. Last question - what happens if you plug AmazonBasics hub to USB2 upstream port? Can you actually turn power off then?

mleguen, 2018-02-18:
That's a good thing you asked that last question, because I discovered my testing device (the one I used to check if the port power was on or off) was not reliable. It has an internal battery, and its charging indicator is not reliable: sometimes it continues to indicate "charging" when it is no longer plugged.

So I replayed all the tests with my smartphone, plugged to the AmazonBasic switch with a power only USB cable (not a data one) to be sure only power on/off can be detected by the phone.

And it works!

Here are the results:
- plugged to an USB2 port: the phone no longer charges as soon as the port is switched off with uhubctl, and charges again as soon as the port is switched back on
- plugged to an USB3 port: the phone no longer charges if and only if the port is switched off on both USB2 and USB3 device, and charges agains as soon as the port is switched on one or both devices

The results are the same is the hub is self powered or usb powered.

I will try to find another testing device (without battery) in the coming days to confirm the results.

mvp, 2018-02-18:
Thank you, @mleguen! I think we can close this for good, I clarified it in eeff46fa0cb.
I will open another issue to implement support for automatic state change for both USB2 and USB3 hubs.

closed 2018-02-18 by mvp

LoganLechat, 2018-02-23:
Hi All, 
I just tested the Amazon basic HUB 7 Port USB3, and i can confirm that work if you connect USB2 upstream port.  In my case i need to repeat few time the power off for my device with this command : 
./uhubctl -l 1-4.3 -p 2 -a off -w 400 -r 5 or /uhubctl -l 1-4.3 -p 2 -a off -w 800 -r 4. 

I just order 2 Amazon Basic HUB 7 and 10 ports USB2 only, and keep in touch if work. 

Thanks

mvp, 2018-02-23:
Thank you @LoganLechat! Information about necessity to add -r 4 is quite helpful, perhaps this should be automated - uhubctl should keep trying to turn off few times until it actually is off.

Do you have access or ability to plug it to USB3 upstream port, and test for USB2 and/or USB3 devices connected to the hub? Be sure to turn off port for both USB2 and USB3 virtual hubs. Thanks a lot!

LoganLechat, 2018-02-24:
Hi MVP. 
To resume my tests : 

- Upstream USB3  + USB2 device = i can power off the device but i need to send 2 power off cmd one to the USB3 hub and the USB2 hub (only the USB2 see my device).
If i send the power off CMD only on the USB2 or only on the USB3 nothing happens power level.

- Upstream USB3 + USB3 device = Power off cmd sent to the USB3 hub, my device restart and reconnect in USB2. If i send again the power off cmd to the USB2 hub the support shutdown normally. 
If you send power ON cmd to USB2 + USB3 the device will keep the USB2 speed. 
If you send power ON cmd to USB3 + USB2 the device will keep the USB3 speed.

- Upstream USB3 **But with USB2 extender cable + USB3 AtoB or USB2 cable AtoB instead USB3 cable AtoB.
The Amazon hub work only in USB2 and the power cmd work perfectly. 

Do you want others tests ? 
Thanks

mvp, 2018-02-24:
Thank you @LoganLechat! That is very helpful for implementing issue #72 properly. Main information here is to turn off USB3 first for USB2 device.