mvp/uhubctl #69: Windows port trial (using usbdk) failed

Manel3, 2018-02-14:
Hi,

I know currently windows port do not work because according to Microsoft winusb.sys does not support necessary USB control requests.

Thank to your suggestion I have found that libusb from 1.0.21 (stable) has usbdk support and I have tried it. The support is not dynamic, you have to recompile libusb with the proper configure flag.

Unfortunately that did not worked, and I opened and issue to usbdk:
https://github.com/daynix/UsbDk/issues/55
And I also opened a issue to libusb:
https://github.com/libusb/libusb/issues/391

In libusb suggested me to open an issue to you, I do not think this issue is yours, but I want to keep you informed.

mvp, 2018-02-14:
@Manel3, thanks for raising it, I'd love to make uhubctl work on Windows. BTW, it was me who suggested you to open issue :)

I hope either Microsoft reconsiders changing winusb.sys to support sending necessary control messages via winusb.sys, or somebody comes up with winusb.sys replacement which doesn't have that limitation.

There are few projects that attempt to create alternative USB drivers, usbdk, libusbK, and few others I guess.

To the best of my knowledge, only old 32-bit libusb-win32 version 0.12 supported this ability - I have used it in my Perl script using Device::USB library on Windows XP back in 2008, and it worked for me.

Manel3, 2018-02-16:
Hi,

See my last comment on daynix/UsbDk#55, I do not think they will add the support we need, but their code can be an example on how to write a filter driver to patch winusb.sys or usbhub3.sys adding control transfer so we can have a Windows working uhubctl.

Do you have experience with WDF (Windows Driver Framework)?

With some libusb changes it could be used from uhubctl without any modification.

mvp, 2018-02-16:
I had some experience with WDF, but it was many years ago. I'm open to any suggestions how to make uhubctl work on Windows, but I'm not likely to start full on hackaton to create such driver for Windows.

As far as I understand, biggest challenge with this would be not so much creating necessary kernel (KMDF) driver, but actually having it signed by Microsoft. Windows 7 and later do not support loading unsigned kernel drivers, not without setting host to debug mode, which is extremely painful process.

mvp, 2018-02-18:
I don't see anything actionable from uhubctl standpoint here.

uhuctl already builds and runs on Windows, it just cannot send necessary control requests to hub to change port power state, and thus power control doesn't work.

If and when `libusb` implements ability to send control messages directly to hubs on Windows (using different version of `winusb.sys` or something else), this issue will fix itself.

closed 2018-02-18 by mvp

YanVugenfirer, 2018-02-19:
* If you want to write your own filter, you can definitely use USBDK as a base. It is distributed under Apache license.

* Signing is not that complicated. You need to get code signing certificate and for Windows 10 EV certificate. No need to certify the driver with WHQL in case of USB filter for the driver to be loaded.

* If you want to extend USBDK functionality, we are open for the patches.

Manel3, 2018-02-19:
Hi,
From what I have found it is needed a KMDF lower filter and MS has an example (upper filter) called kbfiltr, I am going to try to write something (this would be my first driver), it will take me some time.

I see two approaches:
A) specific power control version with specific IOCTL to power on and off the ports of USB HUBS.
B) generic IOCTL to allow any control transfer on any device.

"A" is going to be more easy, and I think I should try starting like this.
"B" is more complicated, but it would be more usefull for libusb users.

USBDK has already a filter on every HUB, that does easy "A" approach, and more difficult "B", but I think that twisting a complex working project to a new not thought on start functionality is going to give a lot of unexpected work.

mvp, 2018-02-19:
@Manel3, as long as you can make this USB control request to succeed (or provide equivalent alternative to it), we're good:

https://github.com/mvp/uhubctl/blob/eeff46fa0cbdefa8ccf91495cddcb1bf60d0c5aa/uhubctl.c#L761-L765