mvp/uhubctl #74: Amazon basics 4 port hub works initially but stops working later on

manuvm317, 2018-03-01:
I am using Amazonbasics 4 port hub connected to a 2.0USB port. The hub worked perfectly fine, I was able to on and off individual ports when I initially used it, but it started breaking after few usages. The status from ./uhubctl shows that the port is off but the device is still charging. Each time I remove and connect to a new PC or a new USB port it works for sometime and then snaps.

mvp, 2018-03-01:
Are you using AmazonBasics 4 port USB 3.0 hub from supported devices list?

Perhaps this is hardware issue with this hub model, e.g. switching circuit not working properly after some use?

Can you try connecting to USB3 upstream port? Remember you will need to turn off ports on both USB2 and USB3 virtual hubs.

manuvm317, 2018-03-01:
Yes using the same AmazonBasics 4 port from the supported devices list.

Yes I tried with the USB3 port as well. The hub does not respond to commands at all. So trying with 2.0 ports now.

mvp, 2018-03-01:
Note that for USB3 operation you need to build from master branch.

mvp, 2018-03-06:
@manuvm317, did you try to build from master branch and follow USB 3.0 duality note in README?

manuvm317, 2018-03-07:
yes I did try that. Hub doesn't respond at all. I have placed order for DLinkDUB-H4 to see if it works.

mvp, 2018-03-07:
It would be helpful if you explained more what you tried to do. Perhaps put logs on pastebin and link to them here.

mvp, 2018-03-11:
@manuvm317, I need more info to debug this.
Please feel free to reopen this issue if you can add anything that I can look at.

closed 2018-03-11 by mvp