mvp/uhubctl #75: Compile error

pedlino, 2018-03-03:
cc -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v1.8-13-g4d84\" uhubctl.c -o uhubctl -Wl,-z,relro -lusb-1.0
uhubctl.c: In function ‘usb_find_hubs’:
uhubctl.c:558:17: warning: ‘libusb_get_port_path’ is deprecated (declared at /usr/include/libusb-1.0/libusb.h:1358): Use libusb_get_port_numbers instead [-Wdeprecated-declarations]
                 int pcount = libusb_get_port_path(NULL, dev, port_numbers, MAX_HUB_CHAIN);
                 ^

I'm a total newb so I don't know what additional information is needed. I'm running ubuntu 14.04 x64 Help?

mvp, 2018-03-03:
It's just a warning, it should still compile and work.

If you want to avoid warning, you need to get libusb 1.0.16 or later. Newer Ubuntu, e.g. 16.04, should have it by default.

closed 2018-03-03 by mvp

pedlino, 2018-03-03:
Actually no... That is the problem. Running uhubctl in the uhubctl folder returns uhubctl: command not found
I can see the file is there and is executable...

mvp, 2018-03-03:
Did you install libusb? Run `sudo apt-get install libusb-1.0-0-dev` - this is mentioned in README.

You may need to update your Ubuntu, run

    sudo apt-get update
    sudo apt-get upgrade

and then try to compile uhubctl again.
My understanding is that on Ubuntu 14.04 you will get libusb-1.0.17, which should not show this compile warning.

pedlino, 2018-03-03:
Thank you for the quick reply. I had already done all that
But I re-run all steps just to make sure.
I get
make: `uhubctl' is up to date.
and again
uhubctl: command not found
Any other clues?

mvp, 2018-03-03:
If you're running from current directory, use `./uhubctl`, or most likely you'll need `sudo ./uhubctl`. To avoid using sudo on Linux, you need to setup usb use permissions.

pedlino, 2018-03-03:
Thanks. Typical DOS newbie error :) I was not using the ./ because that is not needed under DOS.
Unfortunately although I managed to turn off the power for a specific port that did not cut the power supply (i.e. an USB lamp connected still works)
A USB mouse plugged into the port that uhubctl reports as off, indeed doesn't work. So this works but does not solve my problem. 
Anyway, thank you for creating uhubctl and for your patience!

mvp, 2018-03-05:
Note that if you are using USB 3.0 hub connected to upstream USB3 port, you have to turn off power on both USB2 and USB3 ports to really lose power on that port. I am working on implementation to do this automatically, but it's not done yet.

pedlino, 2018-03-05:
This is on a Dell laptop with two internal USB 2.0 hubs. I was trying to turn off a single port in Hub 2. 
The idea was to be able to switch on/off a USB lamp. If you manage to really cut power (unfortunately off is not the same as no-power) please let me know. I'm available for testing. Thanks!