mvp/uhubctl #76: Targus Hub not working

deepakkapoor624, 2018-03-03:
I am using 3.0 Targus Hub but on running uhubctl it shows no smart hub found.

mvp, 2018-03-03:
Targus USB 3.0 hub is not on supported list. You need hub which matches model from the list exactly.

mvp, 2018-03-05:
Closing. Please provide more information if you think that your hub is on supported list.

closed 2018-03-05 by mvp

deepakkapoor624, 2018-03-06:
what additional information you need . Please let me know.

mvp, 2018-03-06:
Just find/buy hub from supported devices list, it should work.

If uhubctl shows no compatible devices detected, your hub isn't compatible. However, on Linux you may need to run it under sudo (or configure udev USB permissions).

deepakkapoor624, 2018-03-07:
I bought Targus usb3.0 hub which is on the supported list . But it still shows not compatible.
What extra thing you have to look for supported device. Thanks in advance.

mvp, 2018-03-07:
There is no Targus USB 3.0 hub on the list, what makes you think so?

Please provide output from uhubctl for your hub, so this can be confirmed.

deepakkapoor624, 2018-03-07:
Targus PAUH212U Hub is mentioned on the supported device list.
I also bought the usb hub of manufacturer Targus with product and vendor Id of 
05e3:0610

mvp, 2018-03-07:
Correct, Targus PAUH212U is on the list, but it is old USB 2.0 hub, not USB 3.0.

You have to find device from the list with exact match - not many vendors make compatible hubs. And there is no guarantee that even if old Targus hub worked that new one would, or other way around - this is question to vendors why they choose to sell devices not conforming to USB standard.

deepakkapoor624, 2018-03-07:
Thanks for your response.