mvp/uhubctl #77: Tag a release

gfa, 2018-03-09:
Hello

I'm about to package uhubctl for debian, would you tag a release so we don't ship an snapshot?

thanks

mvp, 2018-03-09:
Oh, nice! Thanks a lot for doing it! I am about to add some USB 3.0 fixes over this weekend. Can you wait until Monday?

gfa, 2018-03-09:
Sure!

mvp, 2018-03-11:
Done in https://github.com/mvp/uhubctl/releases/tag/v2.0.0

closed 2018-03-11 by mvp

gfa, 2018-03-11:
Cool, thanks

If you use gpg, it would be great if you can start pushing signed git tags

mvp, 2018-03-12:
Thanks @gfa, I will consider using gpg.
Do you want to merge debian specific changes you did in https://salsa.debian.org/debian/uhubctl/commits/debian/unstable back into uhubctl repo?
I don't know anything about debian release process.
Can I make it easier for further releases?

gfa, 2018-03-22:
The only change that would make sense to merge back to your repo is 
https://salsa.debian.org/debian/uhubctl/commit/a60928df05461916f6612f9013c9bedadca5d9cb
but after being reworked as it is debian-specific
This change allows to inject uhubctl's version in the build without having the git repo

Other changes are debian packaging, which are better kept on a different repo than upstream software

Regarding what can you do to make things easier for Debian (and other distros too) 
there is a guide https://wiki.debian.org/UpstreamGuide

however, I must say, you made things pretty easy already! :)

mvp, 2019-09-06:
Note that I have released uhubctl version 2.1.0, can you please update it in Debian? Thanks!

mvp, 2020-05-22:
Note that I have released uhubctl version 2.2.0, can you please update it in Debian? Thanks!

mvp, 2020-06-02:
@gfa , can you please update uhubctl in Debian to version 2.2.0? Thanks

gfa, 2020-06-02:
@mvp i'm working on that

mvp, 2020-12-13:
@gfa, can you please update uhubctl in Debian to version 2.3.0? Thanks

mvp, 2021-02-26:
@gfa, can you please update uhubctl in Debian to version 2.4.0? Note that I do not expect new releases in at least 6 months, possibly more. Thanks!

gfa, 2021-02-26:
Next Debian release is being prepared, while I could package 2.4.0 I think is better to ship 2.3.0 which has been in testing for a while.

As soon as Debian is released, i'll package 2.4.0 and upload it

mvp, 2021-02-26:
Your call, 2.4.0 has slightly better error reporting and suppors toggle action.

mvp, 2021-04-02:
Can you release 2.4.0 now, @gfa?

gfa, 2021-04-05:
uploaded to experimental

mvp, 2024-09-01:
@gfa, can you please update uhubctl in Debian to next version 2.6.0? Release notes are [here](https://github.com/mvp/uhubctl/releases/tag/v2.6.0). Thanks!

mvp, 2024-09-03:
Ping @gfa, can you please release uhubctl 2.6.0 to Debian?