mvp/uhubctl #78: Supported hub

parkerlreed, 2018-03-12:
````
[parker@inspiron15 uhubctl]$ sudo ./uhubctl
Current status for hub 1-3 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.00, 4 ports]
  Port 1: 0100 power
  Port 2: 0000 off
  Port 3: 0303 power lowspeed enable connect [093a:2510 PixArt USB Optical Mouse]
  Port 4: 0503 power highspeed enable connect [0b95:1790 ASIX Elec. AX88179 0000051BA61AF8]
````

The hub itself https://en.j5create.com/products/juh470

parkerlreed, 2018-03-12:
One thing I've noticed: If I have something that's purely power plugged in, it works great. If it's a device plugged in it will turn it off but subsequently turn back on a few seconds later.

mvp, 2018-03-12:
Thanks for your report! Regarding auto reconnection, can you try using option `-r N`, where N is some number, say 10 or even 100. This will try to turn port off many times in succession. Very often this helps to actually turn it off.

mvp, 2018-03-12:
One more question: this is USB3 hub, and are apparently using USB2 connection.

Can you please try connecting it to USB3 upstream port using USB3 compliant cable?

parkerlreed, 2018-03-13:
Aha yeah didn't realize I wasn't using it in the 3.0 port

````
[parker@inspiron15 uhubctl]$ sudo ./uhubctl 
[sudo] password for parker: 
Current status for hub 2-1 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [0b95:1790 ASIX Elec. AX88179 0000051BA61AF8]
Current status for hub 1-1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0303 power lowspeed enable connect [093a:2510 PixArt USB Optical Mouse]
  Port 3: 0303 power lowspeed enable connect [17ef:608c LiteOn Lenovo Calliope USB Keyboard]
  Port 4: 0100 power
````

Normal power off for the mouse didn't stick and spamming occasionally stuck. `-r 100` seems to be more consistent

````
[parker@inspiron15 uhubctl]$ for i in {1..2}; do sudo ./uhubctl -r 100 -a 0 -p 2 -l $i-1; echo "----------------"; done
Current status for hub 2-1 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 00a0 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 00a0 5gbps Rx.Detect
Current status for hub 1-1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0303 power lowspeed enable connect [093a:2510 PixArt USB Optical Mouse]
Sent power off request
New status for hub 1-1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0000 off
----------------
Current status for hub 2-1 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 00a0 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 00a0 5gbps Rx.Detect
Current status for hub 1-1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0000 off
Sent power off request
New status for hub 1-1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0000 off
----------------
````

parkerlreed, 2018-03-13:
Oh is the code automatically trying both the 2.0 and 3.0? I setup the for loop because of the comment in the README lol

parkerlreed, 2018-03-13:
So yeah the code by itself does do both! Nice. Yeah so the rapid succession turn off works.

mvp, 2018-03-13:
Added in fa90f13 - thanks!

mvp, 2018-03-13:
Yes, I have implemented automatic handling for USB3 duality just few days ago.

Note that you can avoid using `sudo` if you configure udev USB permissions.
Simply create a file `/etc/udev/rules.d/52-usb.rules` with following content:

    SUBSYSTEM=="usb", ATTR{idVendor}=="05e3", MODE="0666"

and then run this command to reload udev rules (or simply reboot):

    sudo udevadm trigger --attr-match=subsystem=usb

After doing this, uhubctl will not require root to work.

closed 2018-03-13 by mvp