mvp/uhubctl #80: Anker 14-Port USB 3.0 Data Hub

mdhowle, 2018-03-19:
Anker 14-port (13 data ports) hub appears to work although it can be stubborn.
https://www.amazon.com/Anker-14-Port-Aluminum-Transfer-Charging/dp/B00GSLMTQ8/

Serial Number: 22U2FN2H

```
./uhubctl -a 0 -r 100 -w 30  -l 2-2
```

```
Current status for hub 2-2.4.4.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2.4.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 2-2.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 2-2 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [8564:4000 TS-RDF5 Transcend 000000000037]
  Port 2: 0203 power 5gbps U0 enable connect [8564:4000 TS-RDF5 Transcend 000000000037]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 1-2.4.4.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-2.4.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 1-2.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 1-2 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
```

mvp, 2018-03-19:
We already had another Anker USB 3.0 hub added and then removed in issue #49.

Are you sure that plugged phone stops charging when uhubctl turns off the power for the port?

mdhowle, 2018-03-19:
They are USB SD Card readers. LED next to the port turns off, but when I plug in my phone, it continues to charge but no data connection until the ports are re-enabled.

mvp, 2018-03-19:
That's what I feared... Apparently, Anker hubs lack circuitry to cut off VBUS - saved few cents...

I'm not sure what to do about such hubs. If we add them to supported table, we have to mark that they don't actually work for purpose of turning power off...

mdhowle, 2018-03-20:
uhubctl works for my purpose though I see others' need for full support.  I suppose adding a column stating whether a hub has full or partial (data only) support would be helpful with other with similar use cases as mine.

mvp, 2018-05-19:
I think it's better to not add hubs which don't cut off VBUS to supported table.

closed 2018-05-19 by mvp