mvp/uhubctl #81: Can't run on Ubuntu or centOS

Corwin616, 2018-03-22:
log: 
```
medis@medis:~$ cd Рабочий\ стол/uhubctl-2.0.0/
medis@medis:~/Рабочий стол/uhubctl-2.0.0$ ls
COPYING  Formula  LICENSE  Makefile  README.md  uhubctl.c
medis@medis:~/Рабочий стол/uhubctl-2.0.0$ sudo apt-get install libusb-1.0-0-dev
[sudo] пароль для medis: 
Чтение списков пакетов… Готово
Построение дерева зависимостей       
Чтение информации о состоянии… Готово
Уже установлен пакет libusb-1.0-0-dev самой новой версии (2:1.0.20-1).
Следующий пакет устанавливался автоматически и больше не требуется:
  libllvm4.0
Для его удаления используйте «sudo apt autoremove».
обновлено 0, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 171 пакетов не обновлено.
medis@medis:~/Рабочий стол/uhubctl-2.0.0$ make
fatal: bad revision 'HEAD'
cc -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"\" uhubctl.c -o uhubctl -Wl,-z,relro -lusb-1.0
medis@medis:~/Рабочий стол/uhubctl-2.0.0$ ls
COPYING  Formula  LICENSE  Makefile  README.md  uhubctl  uhubctl.c
medis@medis:~/Рабочий стол/uhubctl-2.0.0$ sh uhubctl
uhubctl: 1: uhubctl: Syntax error: "(" unexpected
medis@medis:~/Рабочий стол/uhubctl-2.0.0$
```

mvp, 2018-03-22:
Your current directory contains spaces, and looks to be Unicode. Can you try running it from directory without spaces, and perhaps using only Latin characters?

I will check why it happens and should provide proper fix anyway, but this should give you a workaround.

mvp, 2018-03-22:
Also, bad revision HEAD suggests you didn't use git to get source code. Current build requires you to git clone.

Corwin616, 2018-03-22:
```
medis@medis:~/app$ git clone https://github.com/mvp/uhubctl.git
Клонирование в «uhubctl»…
remote: Counting objects: 202, done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 202 (delta 17), reused 23 (delta 9), pack-reused 169
Получение объектов: 100% (202/202), 68.05 KiB | 0 bytes/s, готово.
Определение изменений: 100% (111/111), готово.
Проверка соединения… готово.
medis@medis:~/app$ cd uhubctl/
medis@medis:~/app/uhubctl$ ls
COPYING  Formula  LICENSE  Makefile  README.md  uhubctl.c
medis@medis:~/app/uhubctl$ make
cc -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0-2-gfa90\" uhubctl.c -o uhubctl -Wl,-z,relro -lusb-1.0
medis@medis:~/app/uhubctl$ sh uhubctl
uhubctl: 1: uhubctl: Syntax error: "(" unexpected
medis@medis:~/app/uhubctl$
```

mvp, 2018-03-22:
uhubctl isn't a script, but a binary, you can not call it using `sh`. Try `./uhubctl` or `sudo ./uhubctl`.

Corwin616, 2018-03-22:
thanks! it work. soon i will test several hubs in russian stores and report it here

closed 2018-03-22 by Corwin616