mvp/uhubctl #82: List of HUBs with REAL USB power off

bilbolodz, 2018-03-22:
Is it possible to add to list which of compatible USB hubs has function of REAL cut off VBUS power?

mvp, 2018-03-22:
Most old hubs are cutting off VBUS, e.g. D-Link DUB-H7 silver edition. I am going to add one more column to the device table which will state level of support.

In general, I tried to never add hubs to the table which don't cut off VBUS.

bilbolodz, 2018-03-22:
How about newer (actually it should sound: possible to buy nowadays) hubs like: AmazonBasics or some Plugable devices?

mvp, 2018-03-22:
I cannot possibly test all hubs on market, unless someone sends them to me. So, I have to rely on community trying them out and reporting if they work or not.

If you can you test some hubs you have access to and find which ones do cut off VBUS, I would be happy to mention it in supported devices table.

I have to note that while some batches of the same product model do work as expected, but other batches do not work - we've seen it for some AmazonBasics hubs. Perhaps we should add more details than just model id, e.g. mention certain hardware revisions that actually work.

bilbolodz, 2018-03-23:
OK I will do that. I've found one USB 3.0 hub working but unfortunatelly It doesn't cut off power. I will send you a model this evening.

bilbolodz, 2018-03-23:
Working hub BUTwithout cutting off power:
Conrad Electronics UPH340 (2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports). It's 3.0 hub.

mvp, 2018-03-23:
Thanks for report. I will add extra column which states whether VBUS is cut off or not, and add this hubs.

bilbolodz, 2018-03-28:
Something strange: I've D-Link DUB-H7 (Silver) P/N EUBH7E.....A5G H/W ver A5 (2001:f103).
Generally turning off power is working BUT only for some (low power?) devices. I can successfully turn off power when there is NO device connected to hub port. I can also turn off power ("permanently) when to the port is connected on of these device: RS232 converter, STM32 devel board, rtl-sdr donge (RTL2832U), SDRplay (another sdr radio receiver).
When I've connected USB memory stick (checked 3 different) or USB optical mouse I can turn off power (power indeed disappear) but after it about 1 second later port is turned on again (power and USB lines)  so parameter "-a 0" works like "-a 2".

mvp, 2018-03-28:
It is Linux kernel issue - kernel driver(s) notice that device has disappeared, and it tries to power on device again. Usually, this can be fixed by adding option `-r N`, where N is number from 10 to 100. Basically, uhubctl will keep trying to turn off USB port many times in rapid succession.

bilbolodz, 2018-03-28:
Thanks It's working.

closed 2018-03-28 by bilbolodz